arduino/Arduino #5781: http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2 is missing/forbidden

Aerospacesmith, 2016-12-29:
Installing Arduino SAMD Boards 1.6.11 with Windows fails:
Error downloading http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2

Going to http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2 gives me a 403 Forbidden error.

Arduino IDE Log:
```
	Error downloading http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 403
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more
```

facchinm, 2017-01-02:
Probably there was some sort of network error, the link seems up now. Could you please try installing SAMD core again and close the issue if it works? Thanks

NicoHood, 2017-01-02:
I do not know why I see an http link, but it should be changed to https. Its possible in a boards json file.

Aerospacesmith, 2017-01-03:
Thanks NicoHood. Its because the link is a http and not a https.

https://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2 does work. It appears the json file needs updated.

facchinm, 2017-01-04:
The file should be reachable from both http and https (and it IS reachable, I just checked directly and via a proxy). Would you mind testing now (from within the Board Manager) making sure that no warning appears in the Console? Thanks

@NicoHood , we'll also update that link to https but it can't be the root cause of this issue

Aerospacesmith, 2017-01-04:
I believe it has something to do with my work internet. 
I can not access http://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2
But I can access https://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2

mastrolinux, 2017-01-12:
I just tried from many clients referring to that specific server and I can confirm is not an issue on our end. I am going to close the issue but please let us know if the problem appears again. From what I see the 403 forbidden page is not provided by us but from some internal proxy. Please verify with your network administrator.

closed 2017-01-12 by mastrolinux

VladimirAkopyan, 2019-03-05:
This happened again!

per1234, 2019-03-05:
@VladimirAkopyan I just tried downloading the file and had no error at all. It might have been a momentary outage or else you are behind a firewall that is preventing the download.