arduino/Arduino #5783: avrdude has broken dependency to libtinfo.so.5 from ncurses-5

gagara, 2016-12-30:
avrdude from Arduino-1.8.0 has broken dependency to libtinfo.so.5.
```
$ ldd avrdude 
	linux-vdso.so.1 (0x00007ffc2507f000)
	librt.so.1 => /lib64/librt.so.1 (0x00007fcb13f1c000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fcb13d00000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fcb13a04000)
	libtinfo.so.5 => not found
	libc.so.6 => /lib64/libc.so.6 (0x00007fcb1366b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fcb14124000)
```

matthijskooijman, 2016-12-30:
I believe that dependency should be supplied by the system, just like all of the other dependencies that are fulfilled. On Debian systems this file is present in the `libtinfo5` package.

gagara, 2016-12-31:
On gentoo I currently have `ncurses-6.x` installed and `avrdude-6.3` (installed from portage) currently linked to it w/o any problem (i.e. `libtinfo.so.6`).
I'm not sure, but if you are distributing precompiled binaries shouldn't you use static linking in this case (to be sure that they are not depend on system env)?

TheLoko, 2016-12-31:
libtinfo.so.5 is not on Archlinux repos. If arduino needs to use its own avrdude version it should include the correct libraries too.

NicoHood, 2016-12-31:
@TheLoko ArchLinux uses its own avrdude from the upstream sources. Please read the wiki carefully can install arduino-avr-core.

@gagara I think the issue was already in previous versions. You are however right that the libs should be packed static for the binary release of arduino.

facchinm, 2017-01-02:
The problem has reappeared after moving the jenkins slave from Ubuntu 12.04 to Debian 7 in order to fix https://github.com/arduino/Arduino/issues/5430.
Distros packaging `libtinfo6` are usually shipping a compat library to avoid these kind of problems (search for `ncurses5-compat` or similar).
Moving the issue to https://github.com/arduino/avrdude-build-script/issues/4

closed 2017-01-02 by facchinm