arduino/Arduino #5787: access time of reference menu

Myriads, 2017-01-01:
Even though reference contents ara located at the local hight speed SSD, the reference menu of the Arduino IDE takes much longer and sometimes irregular(erratic) than accessing the reference menu of Arduino Home page through Internet. 

Chrome browser used.

Chris--A, 2017-01-02:
Might be best to find the reference in the Arduino install folder. Then open directly in the browser.
As I have html files set to open by default in Notepad++ the link does not even work for me.

per1234, 2022-07-17:
This should no longer occur now that the local reference content has been removed (https://github.com/arduino/Arduino/pull/11495). The menu items now simply open the relevant page from arduino.cc directly.

closed 2022-07-17 by per1234