arduino/Arduino #5792: SD.begin() doesn't work with Adafruit SD Breakout board

basti2k, 2017-01-03:
-   Arduino board: Arduino Mega 2560

-   Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.0

// TESTCODE
#include <SPI.h>
#include <SD.h>

void setup()
{
Serial.begin(57600);
if(!SD.begin(53))
{
Serial.print(F("SD CARD ERROR"));
while(1);
}
Serial.print(F("SD CARD OK"));
while(1);
}

void loop()
{
}

The Code works with the "SD by Arduino, Sparkfun" Lib version 1.0.9 but not with the version 1.1.0

Tested Examples from IDE[1]:
Example | SD 1.0.9 | SD 1.1.0 |
+++++++++++++++++++
Cardinfo | works | works |
Datalogger | works | "Card faild"
Dumpfile | works | "Card faild"
Files | works | "initialization failed!"
listfiles | works | "initialization failed!"
readwrite | works | "initialization failed!"

[1] - Set the CS-PIN to 53

Fede85, 2017-01-03:
Hi,
we have recently updated the SD library to increase its performances but we are receiving some reports like yours.
After some investigation seems that only certain SD cards are affected by this issue. In order to help us troubleshooting can you provide some information about your micro SD such as: brand, model, capacity and class?

Thank you

basti2k, 2017-01-03:
It's a 32G Sony SDHC-Card [Class 10]
[image]

Solder-Soldier, 2017-01-04:
My SD **doesn't** work with 1.8.0   (but does with 1.6.10)
It is a standard size Apacer SDHC 16GB Class 10 formatted with FAT32 
Confirmed that it didn't work today by uploading the same datalogger example from both IDE versions (1.6.1 and 1.8.0)

In 1.6.1 - "card initialized."
In 1.8.0 - "Card failed, or not present"

basti2k, 2017-01-04:
@Solder-Soldier I have installed a old version.
Go under Sketch -> Include Library -> Manage Librarys
Search the SD-Library and select a older Version ;)
[image]

Alain55, 2017-01-04:
I have a similar problem with SD card working with a previous IDE version but not with the 1.8.0 version

    Arduino board: Arduino Mega 2560
    Ethernet board: Arduino
    Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.0

If I use SDfat instead of SD. Sometimes the card initialize normaly, sometimes it won't. I also tried with SD.begin(53, SPI_HALF_SPEED) with the same results.

ncguk, 2017-01-04:
My first experience with using the built-in Arduino SD library in IDE 1.8.0 with an Adafruit Feather Adalogger (and a genuine Arduino Uno r3 with cheap SPI microSD breakout) led to a day of frustration. Swapping from a Samsung 1GB microSD card to a Transcend 4GB card made no difference. Formatting either card as FAT16 or FAT32 (using the official SD card formatter on both Mac and Windows 10) made no difference.

The example sketches would compile but they could not create a new file or read/write from an existing file on the microSD card. In the end I switched to Adafruit's SD library and everything instantly started to work. My conclusion is that the Arduino SD library, as of IDE 1.8.0, is somewhat broken.

sandeepmistry, 2017-01-04:
I've open the following pull request to fix this: https://github.com/arduino-libraries/SD/pull/29 it reverts a commit introduced in https://github.com/arduino-libraries/SD/commit/2f6748f966ba45d1ee5ef0bcab50915ebfe6eec5 which was released with v1.1.0 of the SD library.

basti2k, 2017-01-10:
I tried it today with the version 1.1.1 (of the SD-library).
With my SD-card everything works again :)

sandeepmistry, 2017-01-10:
@basti2k thanks for the update.

This was fixed by https://github.com/arduino-libraries/SD/pull/29, which was merged in the SD v1.1.1 release and is also bundled with IDE v1.8.1.

closed 2017-01-10 by sandeepmistry

ShreyaDBisen, 2017-04-10:
Updating to latest version of SD library helped.
Do try!