arduino/Arduino #5794: Using the SD library to retrieve information over a serial port

dozybe, 2017-01-03:
The sketch in this example is reported as faulty.  SORRY, There is an error at our code repository, please inform to web@arduino.cc

matthijskooijman, 2017-01-03:
Any reason you're not writing to that e-mailadress and instead complaining here? In any case, your report lacks information. You refer to an example, but do not provide the link or otherwise the place where this example is found. You talk about "faulty", but do not specify what's wrong exactly.

dozybe, 2017-01-03:
I'm a newbie. I tried to help. Forget it!

matthijskooijman, 2017-01-03:
I'm not trying to discourage you, reading back my comment perhaps my tone might appear a bit harsh (which was not my intention). In any case, without sufficient information there really is nothing anyone can do to fix this problem. Going on your "forget it", I'll close this issue, but if you want to add more, we can always reopen it if needed.

PaulStoffregen, 2017-01-03:
When you say "The sketch in this example is reported as faulty", can you at least be specific about which sketch or which example?

Yes, you're a beginning and someone probably told you to report a specific problem here.  But regardless of your level of electronics & software knowledge, can you understand how reporting a problem without even telling us which specific file (there are hundreds of files) or web page (hundreds of those too) isn't helpful?  I hope that's just common sense that any beginner can grasp without coding experience.

per1234, 2017-01-12:
Here's the page: https://www.arduino.cc/en/Tutorial/CardInfo
Here's the error message:
>SORRY, There is an error at our code repository, please inform to web@arduino.cc

More pages with the same issue:
https://www.arduino.cc/en/Reference/WiFi101IsAwake
https://www.arduino.cc/en/Reference/WiFi101SleepFor
https://www.arduino.cc/en/Tutorial/Wifi101WiFiSleepingWebClient
https://www.arduino.cc/en/Tutorial/GoogleCalendarWiFi101
https://www.arduino.cc/en/Reference/BLEPeripheralEnd
https://www.arduino.cc/en/Reference/BLECentralConstructor
https://www.arduino.cc/en/Tutorial/TwoPortRXExample
https://www.arduino.cc/en/Tutorial/MouseContinuousDrawing
https://www.arduino.cc/en/Tutorial/GSMExamplesXivelyClient
https://www.arduino.cc/en/Tutorial/GSMExamplesXivelyClientString
https://www.arduino.cc/en/Tutorial/RobotHelloUser
https://www.arduino.cc/en/Tutorial/RobotInputs
https://www.arduino.cc/en/Tutorial/RobotPictureBrowser
https://www.arduino.cc/en/Tutorial/RobotRunawayRobot
https://www.arduino.cc/en/Tutorial/RobotWheelCalibration
https://www.arduino.cc/en/Tutorial/RobotDiscoBot
https://www.arduino.cc/en/Tutorial/RobotCompass
https://www.arduino.cc/en/Tutorial/EsploraTemperatureSensor
https://www.arduino.cc/en/Tutorial/EsploraLedShow2
https://www.arduino.cc/en/Tutorial/EsploraSoundSensor
https://www.arduino.cc/en/Tutorial/EsploraBlink
https://www.arduino.cc/en/Tutorial/EsploraLightCalibrator
https://www.arduino.cc/en/Tutorial/EsploraJoystickMouse
https://www.arduino.cc/en/Tutorial/EsploraMusic
https://www.arduino.cc/en/Tutorial/EsploraLedShow
https://www.arduino.cc/en/Tutorial/CosmClient?from=Tutorial.PachubeClient
https://www.arduino.cc/en/Tutorial/CosmClientString?from=Tutorial.PachubeClientString

>Any reason you're not writing to that e-mailadress and instead complaining here?

@matthijskooijman is that an arduino/Arduino issue tracker policy statement? I realize that emailing website issues to that email address is the recommended approach but the last two emails I sent to that address(some time ago) I received no response. I think an issue tracker is vastly superior to submitting issue reports via email, though some website issue reports are a bit off topic for this repository. In this case these are reference pages for the libraries included with the Arduino IDE, though maybe they would be more appropriate in their respective [arduino-libraries](https://github.com/arduino-libraries) repository? I've seen many issue reports here for website problems and have never seen any mention of them being inappropriate. If that's the case then I'd like to know so that I can be sure to report bugs in the most helpful manner.

facchinm, 2017-01-13:
Looks like a problem with CORS serveer-side strict configuration, we are investigating it

mastrolinux, 2017-01-13:
I just checked, it is not a CORS issue but moved link broken after we moved to arduino-libraries.

Thanks for the extensive report, going to fix them all.

mastrolinux, 2017-01-13:
With some magic I fixed:

https://www.arduino.cc/en/Reference/BLEPeripheralEnd
https://www.arduino.cc/en/Tutorial/GSMExamplesXivelyClientString 
https://www.arduino.cc/en/Tutorial/RobotHelloUser
https://www.arduino.cc/en/Tutorial/RobotInputs
https://www.arduino.cc/en/Tutorial/RobotPictureBrowser
https://www.arduino.cc/en/Tutorial/RobotRunawayRobot
https://www.arduino.cc/en/Tutorial/RobotWheelCalibration
https://www.arduino.cc/en/Tutorial/RobotDiscoBot
https://www.arduino.cc/en/Tutorial/RobotCompass
https://www.arduino.cc/en/Tutorial/EsploraTemperatureSensor
https://www.arduino.cc/en/Tutorial/EsploraLedShow2
https://www.arduino.cc/en/Tutorial/EsploraSoundSensor
https://www.arduino.cc/en/Tutorial/EsploraBlink
https://www.arduino.cc/en/Tutorial/EsploraLightCalibrator
https://www.arduino.cc/en/Tutorial/EsploraJoystickMouse
https://www.arduino.cc/en/Tutorial/EsploraMusic
https://www.arduino.cc/en/Tutorial/EsploraLedShow

Still broken and do not know where to find the proper source:

https://www.arduino.cc/en/Reference/WiFi101IsAwake
https://www.arduino.cc/en/Reference/WiFi101SleepFor
https://www.arduino.cc/en/Tutorial/Wifi101WiFiSleepingWebClient
https://www.arduino.cc/en/Tutorial/GoogleCalendarWiFi101
https://www.arduino.cc/en/Reference/BLECentralConstructor
https://www.arduino.cc/en/Tutorial/TwoPortRXExample
https://www.arduino.cc/en/Tutorial/MouseContinuousDrawing
https://www.arduino.cc/en/Tutorial/GSMExamplesXivelyClient
https://www.arduino.cc/en/Tutorial/CosmClient?from=Tutorial.PachubeClient
https://www.arduino.cc/en/Tutorial/CosmClientString?from=Tutorial.PachubeClientString

@agdl or @facchinm please help me to find the proper tutorials.

per1234, 2017-06-09:
@agdl I think the `Invalid` label should be removed from this issue. The original issue was not described properly but the reported problem was later identified and there are still a significant number of pages that need to be fixed. Even though it's still open, I feel that the `Invalid` label could lead to this issue being overlooked.

SimonePDA, 2018-05-17:
Some tutorials are now here https://github.com/arduino/tutorials , but some other are missing. I'll search and ask who might know. I hope to fix them in a couple of days.

SimonePDA, 2018-05-21:
https://www.arduino.cc/en/Reference/WiFi101IsAwake
https://www.arduino.cc/en/Reference/WiFi101SleepFor
https://www.arduino.cc/en/Tutorial/Wifi101WiFiSleepingWebClient
were pages set up for API that never saw light. Now they are gone.

SimonePDA, 2018-05-21:
https://www.arduino.cc/en/Tutorial/GoogleCalendarWiFi101 - updated
https://www.arduino.cc/en/Reference/BLECentralConstructor - updated
https://www.arduino.cc/en/Tutorial/TwoPortRXExample - updated
https://www.arduino.cc/en/Tutorial/MouseContinuousDrawing - deleted
https://www.arduino.cc/en/Tutorial/GSMExamplesXivelyClient - deleted

SimonePDA, 2018-05-21:
https://www.arduino.cc/en/Tutorial/CosmClient?from=Tutorial.PachubeClient
https://www.arduino.cc/en/Tutorial/CosmClientString?from=Tutorial.PachubeClientString
these two are not linked from anywhere as Tutorials, therefore are obsolete. Deleted

closed 2018-05-21 by SimonePDA