arduino/Arduino #5797: Dangerous bug in Leonardo : USB (serial) logging freezes the device unless buffer is emptied.

AndKe, 2017-01-03:
Try this simple code:(on any leonardo)
I have found Leonardo to fail, and same microcontroller, on a pro micro to work:
```
 unsigned long last;

 void setup() {
   Serial.begin(115200);
   pinMode(13, OUTPUT);
 }

 void loop() {
   if (millis() - last > 999)  {
     last = millis();
     Serial.println("one second");
     digitalWrite(13, !digitalRead(13));
   }
 }
```

 Observe how LED stops flashing whenever you open/close the terminal
 (Ctrl-Shift-M)
 now imagine a Leonardo being used for anything that controls power,
 heat, anything critical - and the logging destination goes offline.

 To put it simply: if the log target is not ready, (DTR ?) - it all goes
 to h*ll

facchinm, 2017-01-04:
Serial over USB (CDC) doesn't have any timeout on AVR. We solved the issue on SAMD with https://github.com/arduino/ArduinoCore-samd/pull/154 , we could use the same approach here unless the current one is the expected behaviour (and I'm sure I've seen at least a project using this start/stop "feature" to control the sketch execution).
@cmaglie @sandeepmistry any thought on that?

NicoHood, 2017-01-04:
So to summarize:
The program blocks because the usb send function blocks. Usb send is executed even though DTR is set (why?). This happens here:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/CDC.cpp#L225

The question is which part of the usb core blocks the sending. Because avr works a bit different:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.cpp#L265

I am not sure why DTR is left HIGH if the port is closed. I could **not** reproduce this bug. If i close the serial monitor the led continues blinking. I'd rather [fix your os](http://getgnulinux.org/en/home/) ;)

AndKe, 2017-01-04:
the OS is Linux , the IDE is current version. And I can only reproduce it on Leonardo, Pro Micro (same microcontroller) works as expected (does not react to console being open/closed.)

facchinm, 2017-01-04:
@AndKe ,
that's really strange... Are you using the same core on both the boards or Pro Micro uses another core/variant? Which Linux distro are you running?
On vanilla Arch the led blinks whether the serial monitor is open or closed. On AVR cores < 1.6.12 there was a bug in suspend/resume (solved with https://github.com/arduino/Arduino/pull/5135) which was affecting Ubuntu/Mint, maybe it's your case?

AndKe, 2017-01-06:
That was 1.6.13  - now tested on 1.8.0  - and cannot reproduce the problem anymore.

closed 2017-01-06 by AndKe