arduino/Arduino #5798: Add NewPing Library to Library Manager

davidbrenner, 2017-01-03:
https://github.com/thimbleio/NewPing

I modified NewPing to make it compliant with the 1.5 library format so it can be included in the Library Manager.

I have contacted Tim (original author of this library) just now to see if he would like to continue maintain this library. I also gave him a link to this issue, so that he can comment here if he would like to maintain the library himself.

teckel12, 2017-01-03:
I wrote the library, and continue to maintain it. Why would I be asked if I still wanted to maintain it? I don't understand what you're doing here other than to cause confusion. The official distribution and support of NewPing is on Bitbucket. If there's something you would like to add to NewPing, it can be added to the official distribution source. 

So yes, I do intend on continuing to support NewPing as I have been doing for years. If you'd like to fork it, kindly make the fork private so others won't be confused with a possibly older version as the latest version is on Bitbucket. 

Tim Eckel, author of NewPing

davidbrenner, 2017-01-03:
Tim, thanks for your response and I apologize for the confusion. My goal is to include the library in the Arduino library manager to make your library easier to access. Doing that requires it to be hosted on github, as per https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

I was likely too eager to see it included in the Library Manager and should have reached out to you first. Again, I apologize. I will delete my repository if you can add a repository here that qualifies to be included in the Library Manager. If you have no interest in doing this, I will rename my fork to avoid confusion with NewPing.

teckel12, 2017-01-04:
I use BitBucket.  It's unfortunate that Arduino would only support Github.  For NewPing to work with the Arduino library manager, the change that needs to happen is that Arduino needs to be fixed to work on multiple repositories, not just GitHub.  In other words, the problem isn't NewPing, it's Arduino that needs to be fixed.  Once Arduino is fixed so there's not an exclusive partnership with GitHub, I can update NewPing to be compatible with the Arduino library manager.  Forcing library authors to use a particular repository is a terrible idea and THAT is what should be fixed.

teckel12, 2017-01-04:
I opened an issue with Arduino to add support of different repositories instead of just GitHub.

https://github.com/arduino/Arduino/issues/5801

davidbrenner, 2017-01-04:
Tim, agreed that Arduino should support bitbucket. I'll add a comment on that ticket adding my support. I'm not sure exactly what is necessary on Arduino's end to support bitbucket, but I am happy to add write code to make it happen.

One comment for you though, can you add your code to a bitbucket repository via mecurial or git rather than just supporting a zip download? I think that would make support on Arduino's end significantly easier. More specifically, can you make your code available at https://bitbucket.org/teckel12/arduino-new-ping/src

PaulStoffregen, 2017-01-04:
I just followed that link, then clicked the "Source" link.  It won't show me any files.  I try logging in, but no, things can't be that simple.  I get redirected to a sign up process for an Atlassian Cloud account.

I understand some people love bitbucket, but this sort of thing really bugs me.

davidbrenner, 2017-01-04:
> It won't show me any files.

That's because Tim currently hosts the code only on the download page: https://bitbucket.org/teckel12/arduino-new-ping/downloads

It's not necessary to create an account to view a repository on bitbucket.

cmaglie, 2017-02-15:
Support for Bitbucket is now available, BTW I cannot see any source code here: https://bitbucket.org/teckel12/arduino-new-ping/src

@teckel12
It seems that you published the code only as zip archives (in the download section) but the source code is not under git.

cmaglie, 2017-04-26:
I'm closing this for lack of feedback, feel free to reopen once you're ready.

AndreKR, 2017-12-10:
@teckel12 Are you going to add a library.properties file, so that the library can be added to the Library Manager?

teckel12, 2017-12-10:
@AndreKR I wasn't aware that Bitbucket support was added till just now. I don't include the source for NewPing because everyone and their brother was forking it and making the forks public, causing confusion with users trying to find the official working version. So I removed the source till Bitbucket allows no forking or forking to always be private.

It's too easy to lose control over a project these days with people forking and screwing stuff up. Due to this, I may just leave it as a Zip file and require the library to be installed manually, which is super easy anyway as there's a function to do it.

teckel12, 2017-12-10:
I wish NewPing forks to be removed from the Arduino Library Manager. They're older versions (current version is.1.9), unsupported and cause user confusion (I get a TON of support emails caused by bad and unauthorized forks).

The proper method of installation is documented on the official NewPing page.

per1234, 2017-12-10:
@teckel12 there is no library named NewPing in the Library Manager index. The fork was never added and your repository could not be added due to you not having source files in it.

teckel12, 2017-12-10:
@per1234 this is all very confusion with changes made outside of my knowledge.  I'm trying to reel control back for this library.  The source files are in the ZIP, just as the the IDE uses to install a library.

PaulStoffregen, 2017-12-10:
FWIW, I have deleted my fork of NewPing from github.

teckel12, 2017-12-10:
@per1234 I believe I've got things sorted out and the source for the library in the correct format can be found here:  https://bitbucket.org/teckel12/arduino-new-ping/src/

I was just made re-aware of this issue a couple hours ago.  As I've only been an active Github user the last few months so I didn't get notifications about this issue and didn't realize that Bitbucket was added to the Arduino library manager.

per1234, 2017-12-10:
Cool! I'm glad to see this moving forward. I apologize for my misunderstanding.

The only remaining thing that must be done before we can add your library is for you to create a [tag ](https://bitbucket.org/teckel12/arduino-new-ping/downloads/?tab=tags)that matches the [`version` value in your library.properties file](https://bitbucket.org/teckel12/arduino-new-ping/src/655032f339232afb01fe842fcfbf56816d8d5658/library.properties?at=master&fileviewer=file-view-default#library.properties-2).

Info on creating tags on BitBucket here:
https://confluence.atlassian.com/bitbucket/use-repository-tags-321860179.html

You will also need to create a tag for any additional versions you release in the future. They will automatically be picked up by the Library Manager index system shortly after that.

teckel12, 2017-12-11:
@per1234 Got it.  Added the tag "1.9.0" on the latest commit to match the version number.  I appreciate the assistance.

teckel12, 2017-12-12:
This issue can be closed now.

per1234, 2017-12-12:
@teckel12 we still need to wait for cmaglie to add NewPing to the Library Manager index. He goes through and does this for all the outstanding inclusion requests periodically so it won't be too long to wait. At that time cmaglie will close the issue. Shortly after that NewPing will be available for installation via Library Manager.

cmaglie, 2017-12-14:
The library has been added!

closed 2017-12-14 by cmaglie