arduino/Arduino #5799: Connect Timed Out Error downloading http://downloads.arduino.cc/packages/package_index.json

engll, 2017-01-03:
Hi, I installed the IDE the first time, and am seeing the following error messages while trying to get into the Boards Manager.  I can access the URL in my browser with no problem.   I am using ubuntu 14.04, and both arduino-1.8.0 and -1.6.13 give me the same error messages.

-----------------------------
 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketTimeoutException: connect timed out
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 5 more

facchinm, 2017-01-04:
Hi @engll ,
can you reach http://downloads.arduino.cc/packages/package_index.json by clicking the link in a browser? If it works and you still get the error from the IDE it could be a proxy/firewall related problem, and you can try adjusting the values under "Preference/Network" to match your network configuration.

engll, 2017-01-04:
Hi @facchinm ,
Thanks for the help.  Yes, it seems like a proxy related problem.  After matching proxy configuration from my browser to "Preference/Network" a dozen or so boards are showing up and the last one is "AMEL-Tech Boards by replaced by Arrow Boards", but I am still seeing the following error messages.  I do have an Additional Boards Manager URL: http://static.dev.sifive.com/bsp/arduino/package_sifive_index.json which I can reach with no problem in a browser.

----------------------------------------------------
 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
        at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
        at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
        at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
        at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
        ... 1 more
Caused by: java.net.UnknownHostException: [Name of Proxy Removed]
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
        at java.net.Socket.connect(Socket.java:589)
        at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
        at sun.net.www.http.HttpClient$1.run(HttpClient.java:484)
        at sun.net.www.http.HttpClient$1.run(HttpClient.java:482)
        at java.security.AccessController.doPrivileged(Native Method)
        at sun.net.www.http.HttpClient.privilegedOpenServer(HttpClient.java:481)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:522)
        at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
        at sun.net.www.http.HttpClient.New(HttpClient.java:308)
        at sun.net.www.http.HttpClient.New(HttpClient.java:326)
        at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
        at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
        at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
        at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
        ... 5 more

facchinm, 2017-01-05:
The new error still seems related to the main package_index, not to the additional one. Maybe your browser is also whitelisted in a local firewall or similar, while Arduino IDE is not? If you try installing a board from Board Manager, does it work or it throws another exception?

engll, 2017-01-05:
I can successfully install Intel i586 Boards .  When I remove the Additional Boards Manger URL, the error messages would not appear.

kb-, 2017-06-16:
I got the same error. I had to delete the C:\Users\xxx\AppData\Local\Arduino15 folder to make it work again.

keomabrun, 2018-01-06:
I have the same issue on Ubuntu 17.10. I don't have any firewall or proxy enabled. Removing the files does not solve the issue.