arduino/Arduino #5801: Arduino library manager needs to support multiple repositories

teckel12, 2017-01-04:
The Arduino library manager only supports GitHub.  Many libraries use different repositories (mine all use BitBucket because I feel it's a far superior repository platform with far superior readme, wiki, and download control.  But, Arduino only supports GitHub for the library manager forcing libraries to either not be included or to create multiple repositories and therefore double the effort.

Instead, Arduino should be fixed to allow for any repository to be supported for the library manager.

davidbrenner, 2017-01-04:
I just want to echo my support. Many arduino libraries out there are distributed through various different means. I would like to see support both for bitbucket and an additional, more generic, means. ie: Any git, mecurial, svn, RCS, or other repository should be supported.

Separate to this, I would also like to see support for .zip distributions (as some libraries are currently only distributed this way). I'm happy to help contribute whatever code is necessary to support this. Just please point me in the direction of the requirements for such a system and I will work on it. 

Thanks!

roncapat, 2017-01-13:
Well, I think that every major online git repo should be supported too... Not every git repo (no self-hosted) but only things such as bitbucket and gitlab for example. So I echo my support for this feature request too.

davidbrenner, 2017-01-13:
Can a maintainer comment on what the next steps are for implementing this request?

Is there a roadmap somewhere for the library manager?

cmaglie, 2017-02-03:
The support for other public online repositories (like Bitbucket) is in testing, it will deployed soon.

cmaglie, 2017-02-15:
Ok, it has taken a while to test, but the support for bitbucket and gitlab is now available.

I've update the FAQ accordingly: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2017-02-15 by cmaglie