arduino/Arduino #5806: https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard unclear which programmer to use and how to replace bootloader for crystal.

codytheking, 2017-01-04:
It is not clear that when burning the bootloader that you choose the programmer under tools as "Arduino as ISP." This should be it's own bullet point before the last step of choosing "Burn Bootloader" under "Tools."

It is also never stated that if the ATmega328 already has a bootloader that requires a crystal, you cannot burn a bootloader to use the internal oscillator without an extra crystal because the chip will not run without a crystal.

facchinm, 2017-01-05:
The tutorial looks pretty good to me (if you read it entirely). 
+1 for the programmer choice to be highlighted with its own bullet point
-1 for the second observation: it works on a brand-new ATMEGA328, since factory fuses are configured to use the internal 1Mhz oscillator

akash73, 2017-01-05:
Added the own bullet point to the doc. https://www.arduino.cc/en/Tutorial.ArduinoToBreadboard
thanks for notice it.

closed 2017-01-30 by akash73