arduino/Arduino #5808: Please add additional libraries to the download manager

MajicDesigns, 2017-01-05:
Now I have sorted out my first 2 libraries, can I please have the following also added to the download manager list:

https://github.com/MajicDesigns/MD_MSGEQ7
https://github.com/MajicDesigns/MD_Cubo
https://github.com/MajicDesigns/MD_AButton
https://github.com/MajicDesigns/MD_DS1307
https://github.com/MajicDesigns/MD_DS3231
https://github.com/MajicDesigns/MD_KeySwitch
https://github.com/MajicDesigns/MD_REncoder
https://github.com/MajicDesigns/MD_TCS230
https://github.com/MajicDesigns/MD_LM335A
https://github.com/MajicDesigns/MD_TTT

Thanks you.

per1234, 2017-01-05:
>https://github.com/MajicDesigns/MD_TTT

Page not found. Did you mean:
https://github.com/MajicDesigns/MD_TicTacToe

MajicDesigns, 2017-01-05:
Yes, my mistake. TicTacToe is the right repository name.  TTT is the library name.

cmaglie, 2017-01-08:
Done, thank you!

closed 2017-01-08 by cmaglie

oly99, 2018-01-26:
can any ne fill me on on why i get 

WARNING: Category 'Game Logic' in library MD_TTT is not valid. Setting to 'Uncategorized'
D:\Downloads\arduino_dht_esp8266_mqtt_ino\arduino_dht_esp8266_mqtt_ino.ino:2:27: fatal error: WiFiEspClient.h: No such file or directory

what is md_ttt its not in any of the scetches why is it asking for it

matthijskooijman, 2018-01-26:
@oly99, your post is completely unrelated to the issue you responded to, so it makes no sense here. Additionally, this issue tracker is for reporting bugs and requesting features in the Arduino IDE, *not* for getting help with your own projects. Please seek help elsewhere, such as on the Arduino forums.