arduino/Arduino #5811: error when try to save a new file

shiftleftplusone, 2017-01-06:
There is an error when I try to save a new file:
for file/new there is no ".ino" appended and the IDE oretends that there already is a file which is called identically, nonetheless there is no such a file or directory actually.
Additionally there is no questeion about optionally "overwrite yes / no ?" 
Appending then manually ".ino" to the suggested file name the works at least finally.
(IDE 1.6.12)
[image]

per1234, 2017-01-06:
You're using an outdated version of the Arduino IDE. I believe this issue was fixed some time ago. Try updating to the latest IDE version.

Please always test using the hourly build from:
https://www.arduino.cc/en/Main/Software#hourly
before reporting issues to be sure they haven't already been fixed.

shiftleftplusone, 2017-01-07:
the most current version is always used  to be bugged and I didn't observe  that this issue has ever  been published so far for either version.

per1234, 2017-01-07:
>the most current version is always used to be bugged 

Just do a quick test with the hourly build to make sure the bug you find still exists before reporting it. I'm not saying you have to actually use the hourly build beyond that. If you're not willing to make that effort then your issue reports won't be a positive contribution to the project.
>I didn't observe that this issue has ever been reported so far for either version.

Here it is: https://github.com/arduino/Arduino/issues/5423. Fixed 3 months and two releases ago.

shiftleftplusone, 2017-01-08:
how should I test without downoloading and installing and reinstalling all my libs and settings to it plus  extra boards just to test and without overwiting and perhaps corrupting my current version ?

per1234, 2017-01-08:
Unless you're running your IDE in portable mode there is no need to reinstall libs or boards or redo your settings. If you were running in portable mode you would only need to copy the portable folder over. In this case the issue has nothing to do with libraries or boards or settings so you wouldn't need to worry about that regardless.

This is all you need to do to test for your issue in the hourly build:
1. Download the hourly build for your OS from https://www.arduino.cc/en/Main/Software#hourly (I'm assuming Windows from your screenshot).
1. Unzip the downloaded file. A folder named arduino-nightly will appear
1. Run arduino-nightly\arduino,exe
1. File > New
1. File > Save As > Save
1. Do you encounter an incorrect "The sketch already contains a file named..." message?
  -**YES**: The bug still occurs. Search the issue tracker to see if issue has already been reported. If not then please report it, including a full description of the problem and steps to reproduce it. You have now made a positive contribution to the Arduino project, thanks!
  -**NO**: The bug has been fixed, no issue report is necessary.
1. Exit the Arduino IDE.
1. Delete the downloaded zip file and the arduino-nightly folder. Alternatively you can continue to use the hourly builds and be a beta tester, reporting any bugs that have been introduced by development. It's possible it may be less stable than the release version but if you encounter an issue that just provides you the opportunity to report it before it ends up being released, much better than the bug being found only after thousands of people are affected.

shiftleftplusone, 2017-01-09:
1st, I have no idea what a portable mode is. I always install by Windows setup.exe. 
for the "save as issue" you're perhaps right, but what if issues only happen just with the complete package containing all my proprietary libs, boards, and settings?
So stop arguing, expecting me to do all that download and reinstall and uninstall mess before reporting issues is completely rediculous. Just simply replying  "Thank you erymuch for your report which is highly appreciated, this issue meanwhile had been resolved  since 1.6.13 (or whatever)" in this case (and all future cases eventually) had cleared all ambiguities and uncertainties without waisting megabytes of serverspace for superfluous chatter and quarreling.

facchinm, 2017-01-09:
@shiftleftplusone a portable installation is self-contained, so it doesn't use any external file outside the IDE directory (so you can have multiple IDEs "installed" ). If you download the zip and create a folder called "portable" inside `arduino-1.x.x` it won't mess with any of your other installations.
Reports are always appreciated but they shouldn't be filed against already solved bugs (if the bug is solved in nightly and not already released it is ok to report it, but since the fix for this bug is already shipped in 2 official releases it is pretty useless and time consuming to report it again).
Thanks

closed 2017-01-09 by facchinm

shiftleftplusone, 2017-01-17:
if there was a search function which successfully found a gtihub result by  keywords
error  save  new file "sketch already contains a file"
then of course I had not reported this TOP issue.
But as already stated,  I will never download any hourly build of perhaps even diverging releases and above all: then also install all my used boards to be able to try to compile.