arduino/Arduino #5812: Please add new library to download manager

lperez31, 2017-01-06:
v1.0.0 of library to connect to YouMadeIt! service is here:
[https://github.com/lperez31/youmadeit-arduino/tree/1.0.0](https://github.com/lperez31/youmadeit-arduino/tree/1.0.0)

Thanks.

Chris--A, 2017-01-07:
You have copied the ArduinoJson library into your library.

This library is already in the library manager: https://github.com/arduino/Arduino/issues/3703

lperez31, 2017-01-07:
Sorry, I don't find in the documentation how I can specify ArduinoJson as a requirement. How can I achieve this?

Thanks.

lperez31, 2017-01-07:
I removed ArduinoJson code from my library. Can you please add it to the library manager?

https://github.com/lperez31/youmadeit-arduino/releases/tag/1.0.1

Thanks again.

Chris--A, 2017-01-07:
I was just pointing it out.
You do not have to remove it, but there could be issues if someone is using the ArduinoJson lib and yours.
You could simply add it as a requirement to your documentation.

lperez31, 2017-01-07:
OK, thanks for pointing it out. I added the requirement in the documentation.

cmaglie, 2017-01-08:
Done!

closed 2017-01-08 by cmaglie