arduino/Arduino #5814: very small IDE at screen resulution 3860x2160

BlindDuck, 2017-01-06:
any text unreadable, while the symbols are easy to miss.

[image]

PaulStoffregen, 2017-01-06:
Squint and use File > Preferences to chance the scale.

BlindDuck, 2017-01-06:
changed to 400%
result:
[image]

things not getting better, unfortunately

BlindDuck, 2017-01-07:
However under Windows 10 scaling of the whole IDE is possible!

The following pictures are made with 250% Scaling of Windows and 100%/automatic of the IED:
[image]
[image]
[image]

And with 225% Scaling in Windows and 400% of the IED:
[image]

stapelberg, 2019-09-07:
As a shot into the dark: check your `gtk-font-name` value. I had `gtk-font-name "Bitstream Vera Sans 8"` in my `~/.gtkrc-2.0`. Changing font size from 8 to 12 worked for me.