arduino/Arduino #5821: ArduinoBoardEthernet cant be programmed with AVRISP MKII, there is a timeout. (Arduinio 1.6.13)

jsiei97, 2017-01-08:
# General description

While programming a ArduinoBoardEthernet with a AVRISP mkII it stops before it is done.
I was using Arduino 1.6.13.

I then tried to enable full verbose mode and cut and pasted the avrdude line and executed it directly in a terminal and that works but please note the time that avrdude is using is approx 688s+680s ~ 22 minutes. 

Therefore I think there is a timeout in the GUI that kicks in before avrdude is done. 

# More information

## Output from GUI
```
Sketch uses 21,058 bytes (65%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,052 bytes (51%) of dynamic memory, leaving 996 bytes for local variables. Maximum is 2,048 bytes.
/opt/arduino/arduino-1.6.13/hardware/tools/avr/bin/avrdude -C/opt/arduino/arduino-1.6.13/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -Uflash:w:/tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex:i 

avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/opt/arduino/arduino-1.6.13/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/cj/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 0000B0018028
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.05
         Vtarget         : 5.1 V
         SCK period      : 953.29 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex"
avrdude: writing flash (21058 bytes):

Writing | ########
```

## Output from shell

```
/opt/arduino/arduino-1.6.13/hardware/tools/avr/bin/avrdude -C/opt/arduino/arduino-1.6.13/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -Uflash:w:/tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/opt/arduino/arduino-1.6.13/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/cj/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 0000B0018028
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.05
         Vtarget         : 5.1 V
         SCK period      : 953.29 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as FD
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex"
avrdude: writing flash (21058 bytes):

Writing | ################################################## | 100% 688.56s

avrdude: 21058 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex:
avrdude: input file /tmp/arduino_build_726629/FunTechHouse_RoomTemperature.ino.hex contains 21058 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 680.08s

avrdude: verifying ...
avrdude: 21058 bytes of flash verified

avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D6, L:FF)

avrdude done.  Thank you.
```

# Hardware used:
1. https://www.arduino.cc/en/Main/arduinoBoardEthernet
1. http://www.atmel.com/tools/avrispmkii.aspx

matthijskooijman, 2017-01-08:
> SCK period      : 953.29 us

It seems the default SPI clock for your programmer is very slow. This is the time required for transferring a single bit, so just the 21k of binary data alone would take around 150s (and there's probably quite some overhead of programming commands, I think that there are 4 bytes needed to transfer just 1 byte of program data).

You could try adding something like "-B 10" to the avrdude commandline (IIRC the number is the bitclock in us). I'm not sure what the max value is (though the clock should not be more than a quarter of the CPU clock, IIRC).

I'm not sure where this slow value comes from - it could be that the AVR ISP MKII remembers it from the last time it was used, or perhaps this is just a default in the firmware of the programmer or in avrdude. In general, the programmer definitions in Arduino do not pass the -B option to just use whatever the default is for the programmer, so I'm not sure if this is something we should be fixing on the Arduino side.

jsiei97, 2017-01-08:
I added the -B 10 and it was really fast, 20s.

Then I tried it again with the GUI and it was using the same speed there, so it took 20s, and worked.
So I guess the problem was in the AVRISP and not the GUI.

Maybe it could be nice to have a error message in the GUI, since the was a little bit confusing since the GUI told me "Done uploading" (even thou it was a failure).

But I guess this issue from my side is done, so I'm closing it (since it was the hw). 
Thanks for the fast reply :)

closed 2017-01-08 by jsiei97