arduino/Arduino #5824: Add mainstream support for dual-controller LCDs (or for a bunch of distinct LCDs seen as a bigger LCD).

roncapat, 2017-01-08:
I recently experimented with a 4x40 LCD. It appears that there is no official support for that. I read about LiquidCrystal440 (or Enhanced LiquidCrystal) but that isn't a official Arduino library.
This is a common problem: someone buys a 4*40 screen and then discovers that he needs to use two instances of LiquidCrystal (one for controller, so one for each couple of lines of the LCD). In OOP terms, this is a weird approach: the device should be represented by a single class, support line-wrapping and let the user refer to a single field of char matrixes.

Do you think that mainstream support for dual-controller LCDs (I'm not referring to my idea, but to the specific problem of 4*40 screen) or similar would not be useful?

-**_Now, my (personal) approach to the 4*40 problem._**
I wrote a higher level class (MultiLcd), that should do the trick (with some -probably unavoidable- overhead): user have to initialise LiquidCrystal instances, ".attach" them to a single MultiLcd instance, and then start to use MultiLcd with the same API of LiquidCrystal. This approach is generic, and the MultiLcd class is able to drive an unlimited amount of similar LCDs as a single LCD, in various layouts. Think to a matrix of LCDs (and obviously to a row or column of them too, if needed -the 4*40 display is seen as a column of two LCDs, for example-). The class I've mocked-up can abstract that matrix of LCDs as a single display. It works, and optimisations are possible.
_If you believe this could be interesting, I'll upload the code somewhere when clean enough._

It would be great to receive opinions or considerations about the problem addressed in this issue. Thank you a lot in advance.

facchinm, 2017-01-09:
As a lot of core libraries, `LiquidCrystal` is not intended to be a complete implementation but it targets the most common hardware with the smallest overhead. External libraries should be preferred when dealing with particular hardware, and your `MultiLcd` library will be really useful once included in the Library Manager.
Thanks!

roncapat, 2017-01-09:
Thank you @facchinm for your answer. I'll follow your advice and I'll upload my library once cleaned and improved a bit. 
Should I open a new Issue, add a message here or submit a PR in order to add my library repo once ready?

closed 2017-01-11 by roncapat