arduino/Arduino #5825: Add MakerVision Library to Library Manager

kmckee, 2017-01-08:
https://github.com/MakerVision/ArduinoLibrary

I'm working on a [Scratch for Arduino web-based Arduino IDE](http://app.makervision.io).  I've noticed that a lot of the code that's generated by the Scratch blocks is a bit overwhelming.  I'd like to introduce an Arduino Library to act as an extra layer of abstraction between the Scratch Blocks and the generated code.

Let me know if you need anything else to add it!

Kyle

kmckee, 2017-01-08:
Looks like it's posted.  Thanks!

kmckee, 2017-01-12:
I posted an update, and it looks like it's no longer in the Library Manager.  Do I need to keep this issue open for it to remain in the store?

cmaglie, 2017-01-12:
This library has never been added!
In fact it sounded a bit weird when you closed the issue some days ago saying that it was published :-)

Anyway, now I've added it!

closed 2017-01-12 by cmaglie

kmckee, 2017-01-12:
Thank you!  I must have had a local install in there and confused myself or something. 

Much appreciated!