arduino/Arduino #5827: Package an up-to-date version for Debian

JulienPalard, 2017-01-09:
Look like current deb package has not been updated in years, leading to users use a deprecated version of Arduino, and reporting very old bugs like https://github.com/arduino/Arduino/issues/5822.

Looks like the deb upgrade is blocked by licence clarity: https://github.com/arduino/Arduino/pull/2703

NicoHood, 2017-01-09:
If you split the packages properly, the licenses are clear to me. Look at the archlinux packages and propose your debian package suggestions. I dont think its a license issue, just nobody wants to package the stuff.

maqifrnswa, 2017-01-10:
this new folder layout (post 1.8) is clear. Packaging it is a pain, but it was already done. I think that debian can be updated.

cmaglie, 2017-02-20:
@JulienPalard @maqifrnswa 
do you know who is actually the maintainer of the debian package?

You should probably ask him to update the debian package and not here.

NicoHood, 2017-02-20:
@cmaglie Please sign the sources #5619 so we can safely package arduino.

cmaglie, 2017-02-21:
> Please sign the sources

@NicoHood thanks for the reminder, there is already #5619 to track the issue no need to duplicate here ;-)

PanderMusubi, 2019-07-14:
Please, see also https://bugs.launchpad.net/ubuntu/+source/arduino/+bug/1425677