arduino/Arduino #5830: BUG - "It's impossible to create the sketch" whith filenames beginning with numbers

roncapat, 2017-01-10:
Version 1.8.0 of Arduino IDE on Ubuntu 16.04.

When opening a file, let's say "1abcd.ino":

Dialog 1:
> It's impossible to create the sketch

(but the folder is ok and the file had been moved as stated)

Dialog 2
> It's impossible to open the sketch: "/path/to/the/sketch.ino"

facchinm, 2017-01-13:
Historically the sketch name could never start with a digit (see https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L853). While a few awkward, it is quite accepted. Closing the issue, reopen if I misunderstood your problem

closed 2017-01-13 by facchinm

roncapat, 2017-01-13:
Well, at least couldn't the IDE throw a self-explaining message about naming rules instead of the errors reported above? It would be useful. I think people waste time trying few times to fight this error, because as the error appears with that message, it's seen as a bug and not a feature.

facchinm, 2017-01-16:
+1 for the self-explaining massage, I'm taking a look and see if it's feasible with a small patch (so all translation won't be affected)