arduino/Arduino #5835: [Library Manager] Please add the Library RF Explorer 3G+ IoT

mballesteros-RFExplorer, 2017-01-11:
Hello,

We are working on our RF Explorer 3G+ IoT library, a customizable Spectrum Analyzer for Arduino! Can you please add to the library manager?

https://github.com/RFExplorer/RFExplorer-IoT-for-Arduino 

Thanks.
Manuel Ballesteros
http://www.rf-explorer.com

cmaglie, 2017-01-12:
The library is not in the correct format: you must move everything in the root folder and remove the folder `RFExplorer_3GP_IoT`.
Once done remember to redo the tag `v1.0.1612.1` (delete and make it again) so it points to the right revision.

cmaglie, 2017-01-16:
Closing for now, please re-open once fixed.

mballesteros-RFExplorer, 2017-03-06:
Hello Cmaglie,

We have created new repo as you explained to us. Please, disregard old url. This is new url repository for our RF Explorer 3G+ IoT library

https://github.com/RFExplorer/RFExplorer_3GP_IoT_Arduino 

Do you mind to  add to the library manager?

cmaglie, 2017-03-07:
Hi @mballesteros-RFExplorer 

there is another problem, the `name` field contains the plus char `+` this is not accepted by the IDE and may produce an error similar to #5955.

To fix this:
1. Remove the `+` sign from the name
2. Commit and push
3. Remove the old tag and re-create a new one.

Thanks!

mballesteros-RFExplorer, 2017-03-07:
Hello Cmaglie,

Thanks for your explanation. We fix this problem and create new tag.

Sincerely,

cmaglie, 2017-03-07:
The library is now published, thanks!

closed 2017-03-07 by cmaglie