arduino/Arduino #5838: printf does not work with %.*s

mamama1, 2017-01-11:
Hi

printf("%.*s", 3, chararray); does not output anything (printf redirected to serial).
printf("%.3s", chararray); works.

this is used when printing char arrays that are not null-terminated.
even if i null-terminate the char array, this doesn't work. printf("%.*s", 3, chararray); still doesn't output anything.
everything else using printf seems to work ok.

is this a bug?

thanks

facchinm, 2017-01-12:
Hi @mamama1 ,
which board are you testing with? AVR core `printf` implementation is not complete (see http://www.nongnu.org/avr-libc/user-manual/group__avr__stdio.html#gaa3b98c0d17b35642c0f3e4649092b9f1 ).
Your case could be 
`
The variable width or precision field (an asterisk * symbol) is not realized and will to abort the output.
`
While the one that works is 
`
s The "char *" argument is expected to be a pointer to an array of character type (pointer to a string). Characters from the array are written up to (but not including) a terminating NUL character; if a precision is specified, no more than the number specified are written. If a precision is given, no null character need be present; if the precision is not specified, or is greater than the size of the array, the array must contain a terminating NUL character.
`
Feel free to reopen the issue if you need further information (or if you are not on AVR :smile:  )

closed 2017-01-12 by facchinm

mamama1, 2017-01-12:
Hi @facchinm,
thanks for clarification. I'm on AVR. So this means that on AVR there is no way to use printf with strings than properly terminating them, right?

How are the chances this will ever be implemented?

facchinm, 2017-01-12:
There is a way, `if a precision is specified, no more than the number specified are written. If a precision is given, no null character need be present`, which is `printf("%.3s", chararray)` in your example. No chance it will change in the future, sorry.

mamama1, 2017-01-12:
Aye, sorry forgot about the %.3s way, I remember that worked, I even pointed it out in my opening post :D thanks for your help.

would it be possible for me to implement it and contribute to the avr libc Project or are the maintainers rather slow/picky when accepting commits?

facchinm, 2017-01-12:
Since AVR platform has huge size constraints, it's rather difficult to make your changes being accepted if they make code size bigger. However, the maintainers are quite active, so at least you'll get a response quite soon :wink:

mamama1, 2017-01-12:
OK I'll try when I find some time.. thanks again!