arduino/Arduino #5839: Why does user need to manually pick library version to update to? And why no "update all" button?

electronicsguy, 2017-01-12:
Arduino IDE shows a popup saying some libraries need to be updated. When I goto the library manager, I must manually select the version to update to. This seems unnecessary. The GUI should select the latest version by default for all updatable packages. If someone wants a different version, they could just pick that. 
[image]

Secondly, I don't see any "update all" button. Am I missing that?

facchinm, 2017-01-12:
Hi @electronicsguy ,
the "Update" button will automatically update to the latest version available if you don't click the dropdown. And the "Update all" button is not there for safety reasons (you could have modified a library and accidentally click that button, losing all your changes)

closed 2017-01-12 by facchinm

electronicsguy, 2017-01-12:
Thanks. So if we infact have modified a library, does the update manager do a 3-way merge?

facchinm, 2017-01-12:
Nope, it completely overwrites the modified library, replacing it with the new one. Very dumb but effective :slightly_smiling_face: