arduino/Arduino #5840: interupt PINOUT arduino MEGA is wrong

naghoss, 2017-01-12:
in first page  for example INT4 is pin2 on arduino  but in second page says is pin19 on arduino.....realy pin 19 === INT4....pinout is wrong...is not  it??? 

https://www.arduino.cc/en/uploads/Hacking/PinMap2560big.png
https://www.arduino.cc/en/Reference/AttachInterrupt

naghoss, 2017-01-12:
how can i report it to ARDUINO for Edit it???

matthijskooijman, 2017-01-12:
I think you're confused by the interrupt numbering. "Interrupt 4" in the reference means "Pass 4 to `attachInterrupt`, which unfortunately does *not* correspond to INT4 in the atmega chip reference (`attachInterrupt` does some remapping of its own, presumably for backward compatibility to keep "interrupt 0" and "interrupt 1" on pin 2 & 3).

It might be good to add a remark about this to the reference though. For example, just above the table with pin numbers, add something like:

> Note that in the table below, the interrupt numbers refer to the number to be passed to `attachInterrupt()`. For historical reasons, this numbering does not always correspond directly to the interrupt numbering on the atmega chip (e.g. int.0 correspons to `INT4` on the Atmega2560 chip).

naghoss, 2017-01-12:
so you say INT4 in atmega  chip reference is equal to INT0 on attachInterrupt??
and what about T0 & T1 ,...(Timer Interrupt)? they are same in atmega chip and arduino board?

naghoss, 2017-01-12:
and is a picture that "show" interrupt pin and timer interrupt pin?

naghoss, 2017-01-12:
i got it tanks about your reply
 M.B.S
mobile num :09177610424

oqibidipo, 2017-01-12:
Here is the pin to interrupt mapping:
```
pin	INTx	digitalPinToInterrupt(pin)
		
3  	INT5	1
2  	INT4	0		
18 TX1	INT3	5
19 RX1	INT2	4
20 SDA	INT1	3
21 SCL	INT0	2
```

CSTaylorJr, 2017-02-16:
Given the above pin to interrupt mapping, how come the digitalPinToPCICR(p) macro in pins_arduino.h (under the directory .\Arduino\hardware\arduino\avr\variants\mega) lists the following:

#define digitalPinToPCICR(p)    ( (((p) >= 10) && ((p) <= 13)) || \
                                  (((p) >= 50) && ((p) <= 53)) || \
                                  (((p) >= 62) && ((p) <= 69)) ? (&PCICR) : ((uint8_t *)0) )

Shouldn't it be:

#define digitalPinToPCICR(p)    ( (((p) >= 2) && ((p) <= 3)) || \
                                  (((p) >= 18) && ((p) <= 21)) ? (&PCICR) : ((uint8_t *)0) )

This appears to be a bug?

oqibidipo, 2017-02-17:
No error there (other than the omission of pins 0, 14 and 15*). Pin change interrupts (PCINT) are not the same as INTx. Up to 8 pins share one PCINT, and as the name suggests, it is triggered by any change to any pin that has PCINT enabled. 

\* The digitalPintoPCI* macros were originally part of the SoftwareSerial library, which explains this comment in `variants/mega/pins_arduino.h`
```
// (I've deliberately left out pin mapping to the Hardware USARTs - seems senseless to me)
```

CSTaylorJr, 2017-02-17:
Ah! Got it!

So SoftwareSerial only uses PCINTs? It doesn't use INTx ints?

oqibidipo, 2017-02-17:
That's correct.

CSTaylorJr, 2017-02-17:
Good to know. Thanks! :)

agdl, 2017-07-17:
I added the sentence @matthijskooijman suggested. Since doubts are cleared, can I close this issue?

matthijskooijman, 2017-07-17:
@agdl, I would move my sentence further down, to the "Interrupt numbers" section, since that's the table I'm referring to. It is now above the list of pin numbers that support interrupts, but that list is not confusing, just the table.

agdl, 2017-07-17:
Yes, you are right. Done sorry

matthijskooijman, 2017-07-17:
Looks good. I would also consider removing the following bit:

> However on newer boards with high performances processors like:
> 
>     Due;
>     Zero;
>     MKR1000;
>     101; 
> 
> a one-to-one mapping between the pin and the interrupt number exist, so the syntaxes:
> 
>     attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
>     attachInterrupt(interruptPin, blink, CHANGE); 
> 
> are equivalent. 

Since that could be interpreted as if it is okay to leave out
`digitalPinToInterrupt` on these boards, but we really should not
encourage this. The fact that interrupt and pin numbers are equal on
these boards is already reflected in the table, so no need to spend so
much text on this fact (which is really an implementation detail).

agdl, 2017-07-17:
Ok done!

matthijskooijman, 2017-07-17:
Heh, noticed one more typo in my addition: correspons -> corresponds. Other than that, looks great!

closed 2017-07-17 by matthijskooijman

agdl, 2017-07-17:
Typo fixed

PaulStoffregen, 2017-07-17:
Arduino 101 is described as all its pins supporting interrupts.

However, only pins 2, 5, 7, 8, 10, 11, 12, 13 work with "CHANGE".  Maybe this should be mentioned?

I mention this because it's been a common misunderstanding when people try to use rotary encoders, either with example code or the Encoder library.  Only those 8 pins are compatible with attachInterrupt uses CHANGE mode.

agdl, 2017-07-17:
@PaulStoffregen added to the table thank you!

PaulStoffregen, 2017-07-17:
Any chance you might add text like this to the Using Interrupts section?

Interrupts are not recommended for use with pushbuttons or switches that mechanically "chatter" as the metal surfaces come into contact with each other.  Your ISR function may be rapidly called many times for a single switch change.  Libraries like Bounce2 are better for dealing with mechanical switches.