arduino/Arduino #5845: [Bug] Windows Store version: collect2.exe: error: ld returned 5 exit status

pczekalski, 2017-01-13:
Dear Contributors,

Thank you for your great work. I've found an error that appears in Windows store 1.8.1.0 version but does not appear when using version downloaded from the acrudino website.
When I try to compile sample code for the SSD1331 library  for the 0.96 inch OLDE display (the BMP Example) triggers following error:

> collect2.exe: error: ld returned 5 exit status

Please mind it happens OLNY in the version downloaded through the Windows Store.
The operating system is Windows 7 x64 Pro.

The verbose output of the compiler is provided below:
`Arduino: 1.8.1 (Windows Store 1.8.1.0) (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware -hardware C:\Users\Piotr\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Piotr\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\libraries -libraries C:\Users\Piotr\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino\examples\bmp\bmp.pde
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware -hardware C:\Users\Piotr\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Piotr\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\libraries -libraries C:\Users\Piotr\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino\examples\bmp\bmp.pde
Using board 'uno' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563\sketch\bmp.pde.cpp" -o "nul"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_GFX_Library" "C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563\sketch\bmp.pde.cpp" -o "nul"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino" "C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563\sketch\bmp.pde.cpp" -o "nul"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\libraries\SD\src" "C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563\sketch\bmp.pde.cpp" -o "nul"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\libraries\SD\src" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI\src" "C:\Users\Piotr\AppData\Local\Temp\arduino_build_477563\sketch\bmp.pde.cpp" -o "nul"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\variants\standard" "-IC:\Users\Piotr\Documents\Ardu [... cut ...]
Using library Adafruit_SSD1331_OLED_Driver_Library_for_Arduino at version 1.0.2 in folder: C:\Users\Piotr\Documents\Arduino\libraries\Adafruit_SSD1331_OLED_Driver_Library_for_Arduino 
Using library SD at version 1.1.1 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\libraries\SD 
Using library SPI at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI 
exit status 1
Error compiling for board Arduino/Genuino Uno.`

facchinm, 2017-01-13:
Almost duplicate of https://github.com/arduino/Arduino/issues/2989, but it's the first time it happens on something different from WinXP. Are you sure you are on Win7 and not on Win10?

pczekalski, 2017-01-13:
My God - I'm so sorry - it is Windows 10 x64 of course (as reported in the attached output section). Just corrected. I'm working parallel on 4 machines and somehow it came on hurry. Sorry for the mistake again.

per1234, 2017-01-13:
@facchinm I have seen multiple reports of this issue occurring with the Windows Store version of the Arduino IDE, including IDE versions previous to 1.8.1 that were released there. Every time when they switch to the standard(non-Windows Store) IDE version the problem goes away. I can provide links to those threads if it's helpful.

pczekalski, 2017-01-13:
Well, indeed it seems to be Windows Store version problem only. Regular (manually downloaded distribution) from Adruio website compiles exactly the same file, without any problems. 
I was just wondering - maybe it is some kind of security and permission related problem underneath that affect Windows-Store installed apps? 
I was browsing Event Log for an errors related but no luck :(.

facchinm, 2017-01-13:
@per1234 , if you could provide the references to the threads it would be great! Long paths (or commandlines) are most likely the culprit here, since the standalone version is going go be installed it a much shorter path than `C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.1.0_x64__mdqgnx93n4wtt\`
However, we are still investigating actively this nasty bug

pczekalski, 2017-01-13:
In case you need some more information or I can do the tests, let me know, please.

Regards,

P.

per1234, 2017-01-13:
https://forum.arduino.cc/index.php?topic=310950.msg3075665#msg3075665
https://forum.arduino.cc/index.php?topic=438628
http://forum.arduino.cc/index.php?topic=436497.msg3008403#msg3008403
https://forum.arduino.cc/index.php?topic=443987.msg3056673#msg3056673

HanntXXX, 2017-01-16:
I'm have the same issue.

cmaglie, 2017-01-18:
Can someone check if the build provided here
https://github.com/arduino/Arduino/pull/5863#issuecomment-273129529
may solve the issue?

pczekalski, 2017-01-18:
I can do it, but how do I install it over the Windows Store?
There is no update to 1.8.2.0 in Windows Store available.
And if it is installed manually, the problem does not occur..

facchinm, 2017-01-18:
@pczekalski I'm uploading the appx for testing, it will be live in a couple of minutes.

BugAA, 2017-01-20:
Same problem - empty sketch only with "delay(1)" in loop is enought for the error message : 

Arduino: 1.8.1 (Windows XP), Board: "Arduino/Genuino Uno"
collect2.exe: error: ld returned 5 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

PLEASE HELP

cmaglie, 2017-01-20:
@BugAA and everyone, please try the build here -> https://github.com/arduino/Arduino/pull/5863#issuecomment-273233625

BugAA, 2017-01-20:
Thank you Cmaglie, but I received positive result just now - it is enough to put first line to that "empty sketch" : 

int X1 = 0; int X2 = 0;

Due to this declaration problem disappeared.

BugAA, 2017-02-01:
Problem again, that declaration (int X1 = 0; int X2 = 0;) helpes only in that empty sketch. Others
sketches give same error again and again. Nothing helps as changing the file  "ld.exe"  to older ..... .
not for understanding.

Producer of ARDUINO please HELP.

facchinm, 2017-02-02:
@BugAA , are you using the build in https://github.com/arduino/Arduino/pull/5863#issuecomment-273233625 ?

BugAA, 2017-02-02:
> Komu: arduino/Arduino <Arduino@noreply.github.com>

Thank you Mr. Facchin for your mail,

I read your recomended link, but excuse me I do not understand what should 
be done.

Inspite of it - behaviour of that error is like random. Yesterday a sketch 
gave error and closing PC for 10 second had not efect.
Today the same sketch goes O.K. immediately after start of PC, but after 
approx. 30 min the error comes again. 
It looks that a reason comes from "Temporary" files ???

Luis

---------- Původní zpráva ----------

(comment)
(https://github.com/arduino/Arduino/pull/5863#issuecomment-273233625) ?

(https://github.com/arduino/Arduino/issues/5845#issuecomment-276901972), or 
mute the thread

"=

BugAA, 2017-02-05:
Next good result - reinstallation IDE.

First step - the IDE 1.8.1 which had been installed by Windows to 
C:\Program files\Arduino ... was canceled by the uninstall proces.

Second step - the IDE install file "arduino-1.6.7-windows.zip" 
was unziped to a folder D:\Arduino\...   .  

Third step - the IDE 1.6.7 is running from that D:\Arduino\ "arduino.exe" 
by clik to the file "arduino.exe"  . 

After this reinstallation the Arduino is running without error. 
Even libraries as "id Crystal" or "DHP22" instaled before reinstallation goes O.K. 

I hope it is end of problem.

Luis

webtop, 2017-03-07:
One thing to note, which could be a factor, or may not :) I noticed that during the download, the store app freezes for a few seconds, then says something like "restoring from ..." (too quick to really see). Could this be causing a corrupt exe? I know that TCP will resend the packet if it's lost, but that doesn't mean it can't be corrupt.

May be nothing, may be something.

For me this issue happens randomly. When it happens I compile again and eventually after any number of random tries, it compiles just fine. Just to be clear, this is a compilation issue, nothing to do with uploading to the board.

cmaglie, 2017-03-30:
Fixed on IDE 1.8.2 (and AVR core 1.6.18).

closed 2017-03-30 by cmaglie