arduino/Arduino #5846: Web Editor Error 403 Forbidden

poiu2463, 2017-01-13:
When I try and access the web editor it gives me this error code. I am trying to access on a Chromebook C720 (Which may be my problem anyways). The error shows up specifically when I try to download the plugin. When I click the download plugin button it redirects me to the 403 error page (Screenshot below)
[image]

facchinm, 2017-01-13:
Hi @poiu2463 ,
what is the url in the titlebar when you get the 403? It's missing in the screenshot... :slightly_frowning_face:

poiu2463, 2017-01-17:
http://downloads.arduino.cc/CreateBridgeStable/

poiu2463, 2017-01-17:
Hi, Sorry for the delay. The url is: http://downloads.arduino.cc/CreateBridgeStable/ 
@facchinm

poiu2463, 2017-01-17:
[image]

poiu2463, 2017-01-17:
[image]

poiu2463, 2017-01-17:
[image]

mastrolinux, 2017-01-18:
Hi @poiu2463 We currently do not support Chromebooks and that's why you get that error. We have a working prototype to support Chromebooks but it is not yet released.
I am closing this issue for now but I can assure you we are working on it on the Arduino Create Web Editor project.

closed 2017-01-18 by mastrolinux

mastrolinux, 2017-07-01:
We released a Chromebook app now available on the official store. You can directly go to the Create Editor and try again.