arduino/Arduino #5849: Add "Clear" button to Serial Monitor UI.

rlaferla, 2017-01-14:
Many times, your Arduino is spitting out a ton of data and you want to manually clear the monitor.  The only way currently (AFAIK) is to restart the app.  I'd like to see a "Clear" button on the GUI that when pressed clears the monitor.

I looked at the code and this could be easily added to SerialMonitor or AbstractTextMonitor with some code similar to below:

```
protected JButton clearButton;

protected void onCreateWindow(Container mainPane) {
  // after adding send button
  clearButton = new JButton(tr("Clear"));
  upperPane.add(clearButton);
}

public void onClearCommand(ActionListener listener) {
  textArea.setText("")
}
```

agdl, 2017-01-19:
@rlaferla you are right there is no button, anyway if you click again on the baud rate the Serial monitor is cleared (faster than close/open again the monitor). Anyway on boards like the UNO this means a board reset

cmaglie, 2017-04-10:
Fixed by https://github.com/arduino/Arduino/commit/99006fc5c68115a55b01b796a50d6a904c9b716c

Thank you!

closed 2017-04-10 by cmaglie