arduino/Arduino #5850: Icons and UI text on retina displays

carlosefr, 2017-01-14:
Icons and UI text on Arduino 1.8.1 are upscaled on macOS with retina displays, even though all remaining interface elements (editor, output pane, status bar) look OK.

[image]

Setting the "interface scale" option in preferences seems to trigger the use of "@2x" icons, but all the UI actually becomes twice the size... :(

neauoire, 2019-03-28:
Same here, there's always a little dotted line across the magnifying glass icon.

[image]

neauoire, 2019-03-28:
Do you know where the icons are located? 
I'll go ahead and make new high res ones.

per1234, 2019-03-31:
@neauoire here they are:
https://github.com/arduino/Arduino/tree/master/build/shared/lib/theme

neauoire, 2019-04-02:
Right, so there are already high resolution icons.
The issue is the upscale, there are no build documentation in the README.

I guess I'll have to get into the mailing list.

per1234, 2019-04-02:
@neauoire thanks for taking the time to look into this! I don't know anything about this topic (actually don't own a Retina display or even a Mac), so unfortunately I can't help you with any useful information specific to this issue.

This issue looks possibly relevant: https://github.com/arduino/Arduino/issues/2309

The Arduino IDE build instructions are here:
https://github.com/arduino/Arduino/wiki/Building-Arduino

facchinm, 2019-04-18:
Related report here too https://github.com/arduino/Arduino/issues/8484#issuecomment-464314147