arduino/Arduino #5852: Provide AppImage for Linux

probonopd, 2017-01-14:
Already today, many Linux users download Arduino directly from the Arduino website rather use what comes with the Linux distribution, since the upstream-provided version is up-to-date while the version in the distributions (especially in stable/LTS versions) is outdated more often than not.

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Just one format for all major distributions
- Works out of the box, no installation of runtimes needed
- Optional(!) desktop integration with `appimaged`
- Binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can GPG2-sign your AppImages (inside the file)

[Here is an overview](https://github.com/probonopd/AppImageKit/wiki/AppImages) of projects that are already distributing upstream-provided, official AppImages.

Do you think this would be useful?
I'd be happy to help producing an official Arduino AppImage.

NicoHood, 2017-01-14:
How about providing proper packages for your distribution to get up to date security and bug fixes instead of creating another static binary blob on your system that you have to maintain yourself.

I have no problems with new ideas, but in this case it would be better people stop complaining and start packaging debian packages which most users use. Read up the documentation and just create your first package. Or use archlinux which does have arduino packages.

probonopd, 2017-01-14:
@NicoHood it's not an either-or, and the AppImage does not add downsides in this regard over the downloads already being provided right now on the Arduino download page.

NicoHood, 2017-01-15:
This all really sounds good, this is a really nice idea. However the time spent on that can be used to generate debian packages and everyone would be happy. But I dont think they will implement this if they cant even #5619 sign their sources.

probonopd, 2017-01-15:
If I understand it correctly, Arduino is outdated even in debian sid:
https://packages.debian.org/sid/arduino

Which I think supports my original point that
> the version in the distributions (especially in stable/LTS versions) is outdated more often than not

NicoHood, 2017-01-15:
Thatswhy I said: Put your energy in updating arduino .deb rather than this binary solution. The reason why its such old is that nobody packages it.

hideout, 2017-01-15:
i think probono this is a really nice idea!
I would love to have Arduino appimage cause i am a Arduiono user too!

cheers

probonopd, 2017-01-20:
It's actually very easy to do. On any Ubuntu or debian 64-bit system, run

```
wget -c "https://github.com/probonopd/AppImages/raw/master/recipes/meta/Recipe"
bash Recipe Arduino
```

It will download the IDE from the official website and convert it to an AppImage. For this, it will use [this yml file](https://github.com/probonopd/AppImages/blob/master/recipes/meta/Arduino.yml) that defines the tar.xz to AppImage conversion, very few lines of code.

probonopd, 2017-03-22:
As a proof-of-concept, I am providing an Arduino AppImage for testing with integrated `esp8266/Arduino` core at

https://bintray.com/probono/AppImages/Arduino_IDE/_latestVersion#files

To try it, just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run. Feedback welcome. Some fine-tuning may still be required, although it is running perfectly fine for me on all Linux distributions I have tried so far.

probonopd, 2017-03-22:
As a proof-of-concept, I am providing an Arduino AppImage for testing with integrated `esp8266/Arduino` core at

https://bintray.com/probono/AppImages/Arduino/_latestVersion#files

To try it, just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run. Feedback welcome. Some fine-tuning may still be required, although it is running perfectly fine for me on all Linux distributions I have tried so far. This proof-of-concept uses an outdated hourly build, but it would be easy to produce these for the most recent builds.

probonopd, 2022-05-07:
Thanks to the help of @per1234 this has been achieved for versinon 2 of the Arduino IDE. Thank you very much! :+1:

Symbian9, 2022-05-07:
> This has been done for versinon 2 of the Arduino IDE

Great news, thanks!

JFTR. Arduino IDE `2.0.0` not released yet, as at the moment only `2.0.0-rc6` is available for testing)

- https://www.arduino.cc/en/software