arduino/Arduino #5858: libListSerialsj does not build from source so it cannot run on FreeBSD

bombasticbob, 2017-01-15:
To the best of my knowledge, libListSerialsj does NOT build from source, but downloads as a binary.  As a result it references a non-existent shared library libc.so.6 when Arduino is built on FreeBSD 11, and the IDE exits on error.

On FreeBSD 11.0, the entire build process for 1.8.1 and 1.6.13 both appear to work properly, using openjdk8 (source retrieved from the specific github branches via 'git clone').  However, because BINARY-ONLY FILES are being used for at least one Java library, the resulting program will NOT run.

I have searched "pretty much everywhere" looking for the source for this library, and it does NOT appar to be ANYWHERE on the internet that a normal search engine can find.

It is actually VERY short-sighted to include binary-only libraries this way.  Assuming that the code is open source, the OPTION to build it from source should be there.

Keep in mind that if a shared library was linked against on ONE system, that it may NOT be 'ABI compatible' with ANOTHER system, regardless of whatever name was given to that shared library (in this case, libc.so.6).  This includes Linux distributions, by the way.  Their package managers differ from one distribution to another.  You cannot assume "it will just work".

FreeBSD is an EXCELLENT development platform.  It currently has a port for Arduino 1.06 .  All that would be necessary to build AND run the Arduinio IDE is to make sure that EVERY library builds from source, rather than "just assuming" that a specific shared library (with a particular ABI, I might add) exists on the host system.  This is what 'build from source' is SUPPOSED to avoid.

Similarly, it should be 'build from source' for ALL of the compilers and tools, in order to make it possible to patch them for new CPUs, at least as an option.  I have grown used to building avr-libc avr-gcc and avr-binutils from source, and have patched them for newer CPUs and submitted those patches upstream for inclusion in the past.

I have noticed that many Linux distributions (particularly Debian, Mint, and probably Ubuntu) also are "stuck" at 1.0.5 or 1.0.6 - and perhaps the reason for this is what I'm describing here:  that compilers and tools destined for Linux have been compiled with hard-coded shared library dependencies that are NOT necessarily compatible across all platforms or all versions.

The simple fix would be to (at least OPTIONALLY) 'build from source' for all tools, compilers, Java libraries, and anything ELSE for that matter.  This is actually the MOST stable way to do things.  And all it would require is to include the source for ALL 3rd party libraries that you're already building to be included in the distribution.

bombasticbob, 2017-01-15:
I'd like to point out a few details regarding this specific library:

a) it seems to be called from Platform.java, with 2 API functions

  private native String resolveDeviceAttachedToNative(String serial);
  private native String[] listSerialsNative();

The output of 'resolveDeviceAttachedToNative()' appears to consist of something like this
"vid_pid"

where 'vid' is apparently a vendor ID, and pid apparently a product ID, probably derived from the USB IDs associated with the serial device.  CLARIFICATION ON THIS, PLEASE!  [that way I can write a replacement for this closed-source lib]

The output of 'listSerialsNative' also appears to consist of something like this:
port_name

where 'port' would be the port you open, and 'name' would be its description.  I am not confident this is accurate, however, but I think I may have investigated this same thing once before, and it's starting to look familiar.

Since I can't seem to get the closed-source library to work, and do *NOT* want to resort to running a windows version, I can only guess from the java code as to what is actually expected here.

ANY clarification as to what these 2 functions actually DO would mean I can write my OWN substitute Java library, and then post instructions (on a github site, probably) on how to BYPASS this closed-source library, and run the Arduino IDE on "other platforms besides their favorite linux version" when it comes to POSIX systems.

I mean, how complicated does this library HAVE to be?  And they can continue to support the 'closed source' version for windows and OSX if they have to.  JUST the Linux source, please.  That would do nicely, and I'd patch it for FBSD as needed.

bombasticbob, 2017-01-16:
found 'unofficial source' trees off of parent project 'arduino' directory.  this may get me going, but doesn't satisfy the general requirement of building the ENTIRE project from source.

https://github.com/arduino/libserialport.git
https://github.com/arduino/listSerialPortsC.git

with any luck THESE will work.

I'm still not happy with the fact that no search engine revealed the location of either of these (someone on IRC suggested it), nor was there any indication at ALL in any of the documentation that I might find the source here.  Now that I know about it, HOPEFULLY anyone else reading this issue will be equally inclined to search in the project directory for possible library sources.  Unfortunately THAT list is several pages long...

kevans91, 2017-01-16:
Hi,

This is actually available in the ports tree as devel/listserialc if you're interesting in taking a look at how we did it there.

A relatively recent version of Arduino (1.6.12) is available in the ports tree as devel/arduino16.

Thanks,

Kyle Evans

facchinm, 2017-01-16:
Hi @bombasticbob , @kevans91 did a spectacular work porting all the needed stuff to FreeBSD (and I believe they all have been merged, correct me if I'm wrong). Having a single build system retrieving all the sources and compiling them requires a huge effort (and it's also very error-prone, since every build would ship "different" binaries). We prefer having a bunch of jenkins jobs compiling and assembling the "distribution" instead than a single massive build script. 
I agree that a lot of documentation on the internals is missing but we are slowly populating it. https://github.com/arduino/listSerialPortsC.git is the repo you are looking for and works good on FreeBSD; Linux distributions are still shipping 1.0.6 for licensing reasons but something is moving so hopefully they'll be packaging mainline quite soon.
I'm closing the issue as solved, feel free to reopen if you need further help.

closed 2017-01-16 by facchinm

bombasticbob, 2017-01-16:
thanks for info on the arduino16 port (I had looked for that on freshports but using the search term 'arduino' didn't show it for some reason).  When I did a google search "arduino 16 site:freshports.org" it came up.  Also noticed the date is more recent than my ports collection, so this is pretty new.  Thanks for mentioning it.  yes, I'd use the patch files and build system in the port for certain.  Hopefully it's a full java build and not just 'download a binary' which is what the arduino port does (or used to do).  Yeah, that's not very helpful to getting the 'bleeding edge' version.

That was the most useful information.  Also the port deals with the build tools, etc.  I can work with it now.

kevans91, 2017-01-16:
Hi,

Indeed, we build all of the dependencies natively -- no Linuxolator
required. The two JNI dependencies are direct dependencies of
devel/arduino16, everything else can be traced out from there through
devel/arduino-tools which brings in the other interesting bits (including
ctags and the toolchain).

I intend to update to 1.8.x shortly, I was mostly waiting for differences
to settle as they were moving things from the IDE to arduino-builder and
changing some ctags stuff as well (IIRC).

thanks,

Kyle Evans