arduino/Arduino #5864: IDE 1.8.1 and A6/A7

andersmoll, 2017-01-17:
I have used 1.6.5-r2 for a long time as I have had some problems going to newer version.
With 1.8.1 all was well upgrading.
But now I have stumbled on a strange issue.

Using some of the pins as digital output, among them number 13.
I now found that some of them, including 13, has around 0.4 volts on the pin as LOW!
This is when compiled with 1.8.1.
When compiling with 1.6.5-r2 the result is 0 volt with pin on LOW.

I traced the problem to a place in the program where I have:
  pinMode( 20, INPUT);
  pinMode( 21, INPUT);
the extra analogue pins on Nano.
When I remove these lines the pins work correctly - LOW is 0 volt!

Something wrong with the initialization of A6/A7 in 1.8.1 ?
Thanks 
Anders

shiftleftplusone, 2017-01-18:
on my ARM cpu the definitions are something like
LOW <=> (U < 0.3*Vref) <=> (U < 1V) // Vref =3.3V; for 5V: U< 1.5V
HIGH <=> (U > 0.6*Vref) <=> (U > 2V) // Vref =3.3V; for 5V: U>3V

In all  cases  V=0.4 V will be correctly LOW for either board (CMIIW)

oqibidipo, 2017-01-19:
>  in the program where I have:
pinMode( 20, INPUT);
pinMode( 21, INPUT);
the extra analogue pins on Nano.

Why? Those pins are **analog only**, there are no corresponding digital pins, so using pinMode is totally pointless and in fact potentially harmful as you have seen. It ends up getting I/O register information outside the bounds of 20-element arrays.

The different IDE versions (or different AVR cores to be exact) use different compiler versions with different options and place those arrays in different order, which explains the different behavior.

andersmoll, 2017-01-19:
OK, I wasn't aware of that pinMode was not required in this case, and even potentially harmful!
I will add code to exclude them from initialization and test.
Thanks a lot!
mvh/anders

closed 2017-01-19 by andersmoll

oqibidipo, 2017-01-19:
It is not _required_ in any case, the digital pins are inputs after reset.

shiftleftplusone, 2017-01-19:
ok, I see, on my Arduinos (Mega, Due)  pin 20 is digital, too.
Very confusing the pin numbering indeed.