arduino/Arduino #5865: documentation missing: set i2c speed on ARM (Due) to 400kHz (FAST-i2c)

shiftleftplusone, 2017-01-17:
CMIIW, but IMO there is a documentation missing about 
set i2c speed on ARM (Due) to 400kHz (FAST-i2c, i2c slave a/o master)

shiftleftplusone, 2017-01-18:
so as this website
http://playground.arduino.cc/Main/WireLibraryDetailedReference
refers only to ATMega and perhaps dofferent AVRs - is everything what's written here also true for the ARM MCUs or not? 
or which is true, and which not?

facchinm, 2017-01-18:
Playground is user-generated content, so you shouldn't take it as reference; the actual Wire documentation is https://www.arduino.cc/en/Reference/Wire, which applies to both AVR and ARM. The I2C speed can be changed using `Wire.setClock()` which unfortunately is not documented on that page. I'll change the issue title to underline that we need to fix the Reference.

akash73, 2017-01-18:
@shiftleftplusone thanks for notice.
Reference page added 
https://www.arduino.cc/en/Reference.WireSetClock

closed 2017-01-18 by akash73

shiftleftplusone, 2017-01-18:
thank you for adding the reference page,
Nonertheless, the site
https://www.arduino.cc/en/Reference/Wire
does not load

shiftleftplusone, 2017-01-18:
thank you, now the reference ist listed!