arduino/Arduino #5866: Reset program button

mgameiro, 2017-01-17:
It would be nice to have a button at IDE to reset the program already inside arduino, without upload the code on computer.

Ivan-Perez, 2017-01-19:
Once you are on the serial monitor, you can reset it by changing the baud rate (click on the dropdown and select the same baud rate you are using). That should reset the Arduino.

matthijskooijman, 2017-01-19:
This isn't possible to implement completely portable right now. Most Arduinos are built so that opening the serial port will reset them, but some (Leonardo) need other prodding to reset. Additionally, third party boards might not support resetting at all. How and if a board is reset is not stored in boards.txt, so I don't think this is as trivial to implement as it sounds. To do it properly, boards.txt would have to get a new property "reset_method" or something? Not sure if this is something anyone wants to invest time in, though.

agdl, 2017-01-19:
@Ivan-Perez actually on boards with an USB2Serial bridge like UNO, Zero (Programming Port) if you press the reset button on the board you see again the code from the start, but in boards like MKR1000, Leonardo this is not possible because the USB is re-enumerated so you loose the USB connection since that serial port is busy

mgameiro, 2017-01-21:
Thank you for responding. 
I though that resetting Serial connection would reset program for all Arduino boards, making it easier to do so. If it doesn't, maybe it really is a little more complicated than that. 
Anyway, a button to close and open again Serial connection could also be useful, and even used like this for most cases.

alexceltare2, 2019-04-09:
I also support the idea. A dedicated button on Arduino IDE to send a reset pulse to the target board will definitely be a great idea since there's a lot of room in the buttons area to use.