arduino/Arduino #5867: Ethernet2 - Missing functions

DieterBurandt, 2017-01-17:
The Ethernet2 (and Ethernet) library supports reading the following parameters from the W5500 board:
  - localIP
  - subnetMask
  - gatewayIP
  - dnsServerIP

Curiously the two other parameters are missing:
- MacAddr
- dhcpServerIP

For maintenance purpose the two missing calls should be added. The W5500 driver already supports these calls, so it is very simple to do:

For the new Arduino Ethernet Shield V2 R1.0.3  ==>

Add to Ethernet2.h =>
	IPAddress dhcpServerIP();
	void      MacAddr(uint8_t * Mac);
    
and to Ethernet2.cpp =>
	IPAddress EthernetClass::dhcpServerIP()
	{
	  return _dhcp->getDhcpServerIp();
	}
	void EthernetClass::MacAddr(uint8_t * Mac)
	{
		w5500.getMACAddress(Mac);
	}

per1234, 2018-07-26:
`Ethernet.MACAddress()` has been added by https://github.com/arduino-libraries/Ethernet/commit/ac843291454f522cdef6bddbacce2c52028be943

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/Ethernet/issues/155

closed 2021-03-28 by per1234