arduino/Arduino #5874: BUG: EEPtr++ is incorrectly equivalent to ++EEPtr

eric-wieser, 2017-01-18:
See [these lines](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/EEPROM/src/EEPROM.h#L100-L101).

Post-increment is supposed to return the value before the increment occured.

However, the `,` operator is a sequence point, so despite attempts to the contrary, the result here is that the value after the increment is returned.

Pinging @Chris--A

Chris--A, 2017-01-18:
@eric-wieser Good find. I'll definitely patch this to my changes (https://github.com/arduino/Arduino/compare/master...Chris--A:EEPROM_2.1)

eric-wieser, 2017-01-18:
Actually I'm wrong, and this is all fine

closed 2017-01-18 by eric-wieser

Chris--A, 2017-01-18:
nice. I had not even started to test it. But yeah, I too mixed up the overloads on first glance.