arduino/Arduino #5876: Add the <typetraits> header

eric-wieser, 2017-01-19:
Since all of this part of the STL runs at compile-time, arduino being an embedded platform is not a good reason to exclude it.

In particular, things like `std::remove_reference` are needed to implement #5875

cmaglie, 2017-01-24:
Why you say it's excluded? AFAIK the build of the toolchain did not exclude anything, maybe those are not implemented yet for AVR?

eric-wieser, 2017-01-24:
You're right, this is really a complaint against avr, not against arduino.

It might be nice if arduino could provide it to make up for its absense from avr though

cmaglie, 2017-01-24:
Ok the question is then, where we can find a good implementation of them? I think that Arduino doesn't have enough resources to maintain his own implementation of the standard C++ lib.

matthijskooijman, 2017-01-24:
uclibc++ has a decent implementation (also available in a few Arduino libraries), though it is incomplete (especially most of C++11 is missing). Not sure if `type_traits` is among it.

cmaglie, 2019-09-16:
Closing this as wontfix.

closed 2019-09-16 by cmaglie