arduino/Arduino #5885: Add HardWire Library to Library Manager

thexeno, 2017-01-19:
https://github.com/thexeno/HardWire-Arduino-Library

I have modified the Wire library in order to make it more flexible at low level without increasing the programming complexity, using the AVR architectures. 

It helped me to configure Arduino as I2C slave and handling data at bit or byte level within the START and STOP condition, as many IC would do, increasing the flexibility. Basically manages a low level I2C bus transactions, executing a user defined callback at each phase of the I2C communication. This is made by keeping a full compatibility with the Arduino environment and so making lower level stuff in Arduino based sketches, when needed. It works on projects already based on Wire, so is "backward/wire" compatible and can be used also as a pure Wire, if the additional callbacks provided are not used.

Hope you can find it helpful.

agdl, 2017-01-20:
I'm sorry, but your library can't be added since a tag is missing  please follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)

thexeno, 2017-01-21:
Now it should fixed.

agdl, 2017-01-24:
Done thank you!

closed 2017-01-24 by agdl

thexeno, 2017-01-25:
A comment on the versioning: why it is available in the Library Manager of the Arudino IDE as a version 1.0.0 instead 1.0.1?