arduino/Arduino #5888: Line continuation character ignored by syntax highlighting

septillion-git, 2017-01-20:
The line continuation character \ is not parsed by the syntax highlighting if it's part of a line comment.
```C++
\\ I start a comment on this line but I end with the line continuation character \
byte test = 5; //This WHOLE line is a comment because of the line continuation character on the last line

//The byte test is thereby NOT declared but the syntax highlighting of the IDE does not display it as comment (Markdown syntax highlighing does not either :P )
```

This way it's easy for a newbie to do this without knowing what's wrong. If the line was highlighted ass comment the error would have been more clear.

Link to a case where it happened:
http://forum.arduino.cc/index.php?topic=449738.0

facchinm, 2017-01-20:
The feature you are requesting should be implemented by [RSyntaxTextArea](https://github.com/bobbylight/RSyntaxTextArea/), the Java library we use as text editor. 
Reopening the issue there will surely boost the visibility of this issue. Thanks for reporting!

septillion-git, 2017-01-20:
Thanks for pointing me to RSTA! You talk about a closed (but unfixed) issue over at RSTA, I can't find it, do you have by any change the issue number?

facchinm, 2017-01-23:
Sorry @septillion-git , I meant "file a new issue there about the same topic" :smile: Thanks!

septillion-git, 2017-01-23:
Thanks! I will :)

Quick other question, is it by design that c-style strings are not highlighted anymore? On 1.6.4 they turn blue, on 1.8.1 nothing happens to them. Or is that a RSTA bug as well?

septillion-git, 2017-01-25:
RSTA fixed the bug :) Hopefully it's implemented in the IDE quick .

Still remain the question about strings...

facchinm, 2017-01-25:
As soon as 2.6.1 gets released we'll import it in the IDE! Thanks for pushing this!

commit f5eb673f (2017-03-01): Update RSyntaxTextArea to 2.6.1

This imports the unmodified upstream version of RSTA from
http://repo1.maven.org/maven2/com/fifesoft/rsyntaxtextarea/2.6.1/rsyntaxtextarea-2.6.1.jar
since the Arduino-specific changes are no longer required.

This fixes #5888
closed 2017-03-01 by cmaglie