arduino/Arduino #5891: [website] Ethernet Shield 2 product page issues

per1234, 2017-01-20:
1. https://www.arduino.cc/en/Main/ArduinoEthernetShield has a link
> Need Help?
    On the Ethernet Shield V2 [on the Ethernet V2 Library ](https://www.arduino.cc/en/Reference/EthernetV2)

but the linked page https://www.arduino.cc/en/Reference/EthernetV2 just says:
>The page "Reference/EthernetV2" doesn't exist.

2. The wording of:
>On the Ethernet Shield V2 [on the Ethernet V2 Library ](https://www.arduino.cc/en/Reference/EthernetV2)

is bad. Was that meant to be on two different lines or ?

3. The name of the library has always been Ethernet2, not Ethernet V2. The [library.properties](https://github.com/arduino-org/Arduino/blob/master/libraries/Ethernet2/library.properties#L1) file in the arduino.org IDE repository has the name value set to Ethernet2. Same in the [Adafruit repository](https://github.com/adafruit/Ethernet2/blob/master/library.properties#L1). This is important because [Seeed Studio's library](https://github.com/Seeed-Studio/Ethernet_Shield_W5200) for their W5200 shield is included by the file EthernetV2_0.h so calling the Ethernet2 library by the wrong name will cause even more confusion than already exists for people trying to figure out which library they need to use for their shield.

4. There are multiple references to the Ethernet library on that page. There has been a constant stream of posts on the Arduino forum from people not being able to use their Ethernet Shield 2 with the Ethernet library ever since that shield was released. It's very important not to contribute to this by pointing them to the wrong library.

5. The page has a link to the Ethernet shield Getting Started page:
> Getting Started
In the [Getting Started section](https://www.arduino.cc/en/Guide/ArduinoEthernetShield),

Which is for the wrong product and links to the Ethernet library.

6. The title of the page is "Arduino Ethernet Shield V2" and several other places on the page it is called this but elsewhere else in the page, on the silkscreen, and on arduino.org's product page the shield is called "Arduino Ethernet Shield 2". Calling it "Arduino Ethernet Shield V2" will lead to more confusion with the Seeed Studios library for their W5200 based Ethernet Shield.

7. [Shop Now](https://store.arduino.cc/product/A000024) link leads to a 404 error (as do the buttons on the other newly added product pages).

8. "Do you own a past an old version of this product? Check Arduino Ethernet Shield V1 product page." should be something like:
>Do you own **a past** version of this product? Check **the** Arduino Ethernet Shield V1 product page.

q2dg, 2017-01-22:
Well, this happens when you put on market a product without its corresponding library: it's a shoot on the foot. I'd suggest keep on https://github.com/arduino-libraries/Ethernet/issues/37

frrederik, 2017-11-24:
All outstanding issues have now been corrected.

per1234, 2017-11-24:
Thanks @frrederik!

I'd like to suggest one small adjustment. Change:
>Learn more on the Ethernet Shield 2 [in the Ethernet 2 Library](https://www.arduino.cc/en/Reference/Ethernet)

to:
>Learn more on the Ethernet Shield 2 in the [**Ethernet2** Library **reference**](https://www.arduino.cc/en/Reference/Ethernet)

(correct spelling of the library name and addition of the word "reference")

----
My reference for correct library name:
https://github.com/adafruit/Ethernet2/blob/master/library.properties#L1
```
name=Ethernet2
```

frrederik, 2018-01-24:
Hi @per1234,

I have now committed the changes to the wording and precision in naming. Feel free to close.

thx!

closed 2018-01-24 by per1234