arduino/Arduino #5893: change default UART baudrate: Serial.begin(9600)  far too slow!

shiftleftplusone, 2017-01-21:
please change default UART baudrate: 
Serial.begin(9600)  far too slow!
it very often causes data consumption and data transmission blocking!
Please set the default to 38400 or even 115200, both for the IDE, for the serial monitor defaults, and for all Arduino Sketch examples!

cousteaulecommandant, 2017-01-21:
9600 is somewhat standard, I guess that's why it was chosen, but it's true that it may be too slow.  If this is changed I'd go directly to *115200* (faster than that might be unsupported; I think it's non-standard).  And definitely this should be normalized across the code uploader, the serial monitor/plotter, and the examples.

facchinm, 2017-01-23:
I believe 9600 is still there for historical reasons; @tigoe can confirm this :slightly_smiling_face: 
However the baud rate is far less important in modern boards with `Serial` mapped on USB CDC. I personally prefer 115200 but probably for starters it's easier to remember that 9600 is the "magic number", and will surely work on ALL boards at the cost of a slight performance penalty (that you can avoid once you learn more about baud rates and serial ports)

tigoe, 2017-01-23:
Compatibility is the reason it's still there, yes. And time.

If you want to change it, make sure you change it for every example and
library that Arduino maintains, including the tutorials, and make sure it's
compatible with all currently supported boards. Also check the serial
monitor with a long stress test to make sure it doesn't have problems after
several minutes. Last time I checked, it did. Probably should do the same
for the Create agent. And make sure you coordinate it with IDE releases of
the desktop and Create, so you can change their defaults at the same time.

Changing the whole system over to default 115200 would be significant work
at this point. If you do it, make sure you have the time and personnel to
complete it beforehand, and plan it thoroughly. It would be a mess if it
were left half-finished.

or mute
the thread

pwillard, 2017-07-06:
i see no reason to move away from 9600 baud as a default.  For a whole lot of things... its actually "fast" enough.

sandeepmistry, 2019-09-16:
I'm closing this for now, we don't have any plans to change the default baud rates.

closed 2019-09-16 by sandeepmistry