arduino/Arduino #5894: improvement request: please port Due Scheduler to AVR

shiftleftplusone, 2017-01-21:
hello,
for the Due C. Maglie's and S. Fitzgerald's Scheduler lib 
https://github.com/arduino-libraries/Scheduler/blob/master/examples/MultipleBlinks/MultipleBlinks
is both powerful and handsome - 
can you please port it to work for AVRs, too?

The currently existing bunch of different MT libs is totally weird and absolutely confusing, so why not have 1 lib with 1 syntax that works for all?

mikaelpatel, 2017-01-21:
This Scheduler works with AVR, SAM, and SAMD; https://github.com/mikaelpatel/Arduino-Scheduler
It also has more support classes, higher performance and lower foot-print.

shiftleftplusone, 2017-01-21:
that's fine!
But then it is a documentation issue, because in the reference  it's dedicated just for Due and Zero Boards:
https://www.arduino.cc/en/Reference/Scheduler

> Reference   Language | Libraries | Comparison | Changes
Scheduler

The Scheduler library enables an Arduino based on SAM and SAMD architectures (i.e (i.e Zero, MKRZero, MKR1000 Board) and Due) to run multiple functions at the same 

> time.

shiftleftplusone, 2017-01-21:
yes, perhaps, but that's a different lib which is waaay more complicated!
My topic here is just about  the Due Scheduler
https://github.com/arduino-libraries/Scheduler/blob/master/examples/MultipleBlinks/MultipleBlinks
which should be ported because of it's easy and handsome API usage.

mikaelpatel, 2017-01-21:
@shiftleftplusone If you did read the code and API you will see that it is actually much less complex, if measured with software quality measurement tools. The API is lighter, better documented and the implementation less complex with regard to number of lines, documentation and portablility. 

The Due Scheduler cannot be ported to the AVR. It is not written as portable code (e.g. assembly). Instead it has to be rewritten (with the same API). There have been some attempts to rewrite for AVR but they have not been merged into the official library. I do not know of any plans to do that and if I have understood correctly Arduino Due is discontinued. 

Finally, there has not been any major work on this library since 2012-2013. Please check the commit log; https://github.com/arduino-libraries/Scheduler/commits/master

shiftleftplusone, 2017-01-21:
no, the Due is not discontinued.

But if you pretend that your lib is as quite as simple as the Due Scheduler, how would your syntax look like for this example code insead? Perhaps you might convince me (tbh, I am sceptical)::

`
/*
 Multiple Blinks
 Demonstrates the use of the Scheduler library for the Arduino Due
 Hardware required :
 * LEDs connected to pins 11, 12, and 13
 created 8 Oct 2012
 by Cristian Maglie
 Modified by
 Scott Fitzgerald 19 Oct 2012
 This example code is in the public domain
 http://www.arduino.cc/en/Tutorial/MultipleBlinks
*/

// Include Scheduler since we want to manage multiple tasks.
#include <Scheduler.h>

int led1 = 13;
int led2 = 12;
int led3 = 11;

void setup() {
  Serial.begin(9600);

  // Setup the 3 pins as OUTPUT
  pinMode(led1, OUTPUT);
  pinMode(led2, OUTPUT);
  pinMode(led3, OUTPUT);

  // Add "loop2" and "loop3" to scheduling.
  // "loop" is always started by default.
  Scheduler.startLoop(loop2);
  Scheduler.startLoop(loop3);
}

// Task no.1: blink LED with 1 second delay.
void loop() {
  digitalWrite(led1, HIGH);

  // IMPORTANT:
  // When multiple tasks are running 'delay' passes control to
  // other tasks while waiting and guarantees they get executed.
  delay(1000);

  digitalWrite(led1, LOW);
  delay(1000);
}

// Task no.2: blink LED with 0.1 second delay.
void loop2() {
  digitalWrite(led2, HIGH);
  delay(100);
  digitalWrite(led2, LOW);
  delay(100);
}

// Task no.3: accept commands from Serial port
// '0' turns off LED
// '1' turns on LED
void loop3() {
  if (Serial.available()) {
    char c = Serial.read();
    if (c == '0') {
      digitalWrite(led3, LOW);
      Serial.println("Led turned off!");
    }
    if (c == '1') {
      digitalWrite(led3, HIGH);
      Serial.println("Led turned on!");
    }
  }

  // IMPORTANT:
  // We must call 'yield' at a regular basis to pass
  // control to other tasks.
  yield();
}
`

mikaelpatel, 2017-01-21:
I understand that you are skeptical :). It too a few seconds to get that working (on an Arduino Uno/IDE 1.8.1). Changed:
```
  Scheduler.startLoop(loop2);
  Scheduler.startLoop(loop3);
```
to:
```
  Scheduler.start(NULL, loop2);
  Scheduler.start(NULL, loop3);
```
The setup for respective tasks should be function (and replace the NULL). You could also simple add the following `#define startLoop(fn) start(NULL, fn)`.

Anyway, just trying to help. 

PS: Please ignore the comment on Arduino Due being discontinued. Obviously that has nothing to do with this issue.

SurfingDude, 2017-01-21:
The ESP8266-based boards could make very good use of the Scheduler Library, if it is possible!

shiftleftplusone, 2017-01-22:
thank you mikaelpatil, I didn'texpect that it ws so easy to port the code. I was puzzled by things like (taken from different examples) 
```
#if defined(ARDUINO_ARCH_SAMD) && (USB_PID == 0x8D21)
#define Serial SerialUSB
#endif
 int stack;
 stack = Scheduler.stack();
  Serial.flush();
  int tasks = 0;
  const int TASKS_MAX = NUM_DIGITAL_PINS - 2;
  while (tasks < TASKS_MAX && Scheduler.start(NULL, blink, 64)) tasks++;
// and the NULL in 
  Scheduler.start(NULL, controllerLoop);
  Scheduler.start(NULL, buttonLoop);
```
that were the things which kept me away from starting  to try it.
perhaps I would recommend to shrink that down in the basic user examples not to scare people (just my2ct) , 
but if it actually is that easy as 

> Changed:
>   Scheduler.startLoop(loop2);
>   Scheduler.startLoop(loop3);
> to:
>   Scheduler.start(NULL, loop2);
>   Scheduler.start(NULL, loop3);

and keep all the rest unchanged, then I'll give it a try on my Mega!
Thanks for your advice!

shiftleftplusone, 2017-01-25:
hello 
@all, @ Mikael Patel:
I now tested C. Maglie's Multiple Blink example with your (Mikael Patel's) Scheduler lib, change just by the 
`#define startLoop(fn) start(NULL, fn)`

- and it works, out of the box, both on Arduino Due and Arduino Mega.

Great work, thanks a lot.

(topic may be closed)

cmaglie, 2017-01-25:
@mikaelpatel it may be worth to consider adding a `startLoop` method directly in your library so users may exchange the Scheduler implementation at the only cost of changing the `#include`.

closed 2017-01-25 by cmaglie

mikaelpatel, 2017-01-25:
@cmaglie yes, I was considering an update so that the libraries are directly compatible. On my to-do-list :)

shiftleftplusone, 2017-01-25:
as you just mentioned an update to come:

would you perhaps consider also to enable tasks to be terminated (e.g., by local expressions or even by environmental semaphores)?
Then it would be possible not only to start perpetual tasks but also some which are running just temporarily and optionally may be restarted later again in case of need.

Example:
Creating PID tasks for different motors, running independently and simultaneously until the targeted end condition has been reached, and then terminate, and then can be started for a new PID target.

I just created such a thing using gcc feat. pthread on my Raspberry Pi.

mikaelpatel, 2017-01-25:
@cmaglie @shiftleftplusone Please see the update version (commit https://github.com/mikaelpatel/Arduino-Scheduler/commit/24b11bfa9d00048f4b1c2b7f2c359b6f33776eb6).

mikaelpatel, 2017-01-25:
@shiftleftplusone Please feel free to use the issue list for the library. Quick answer is all that can be done with the available semantics. There are some restrictions as the tasks are (must be) allocated on the stack.

shiftleftplusone, 2017-01-25:
ok, thanks, I just opened a new issue over there!