arduino/Arduino #5895: example code update needed: File / Examples / LiquidCrystal / CustomCharacter

birozso, 2017-01-21:
The purpose of the post is in the SUBJECT line.

It is the update of the example file of arduino IDE.  It does not show the I<3 Arduino sentence of the first row of the LCD. Missing setting position of the LCD writting. So i added this. 

In red you can see my added lines, which correct the original code and make it work.

path of original code, should change:
File / Examples / LiquidCrystal / CustomCharacter
(CustomCharacter.ino)

i hope it is clear . 
[CustomCharacter.zip](https://github.com/arduino/Arduino/files/721077/CustomCharacter.zip)
[image]

regards

per1234, 2017-01-21:
See https://github.com/arduino-libraries/LiquidCrystal/pull/13

facchinm, 2017-01-23:
Closing as duplicate

closed 2017-01-23 by facchinm