arduino/Arduino #5899: Serial.print() appears to overwrite the function call stack

ryan039, 2017-01-22:
Hello, 
I have a simple sketch that is storing a sequence of button pushes into EEPROM memory.
If the button is not pressed fast enough there is a tendency for the board to throw itself back to the setup() function.
I've included Serial output IOT try and troubleshoot in the Serial Monitor, If the button is pressed and held, it should immediately fill up the allotted presses for the code sequence, instead the serial monior outputs gibberish characters and the board loops itself back to setup().

The circuit is super straight forward and easy to build, and the sketch is short:
You can paste the following into your IDE:

`
/*
 *    My intent is to have a person hit a button in a sequence and
 * have the board store that sequence in EEPROM memory. 
 *    On the next startup, the user will enter that sequence and compare it to the "Master" sequence.
 * If they match....launch nukes - otherwise nothing.
 *    I'm storing the sequence in EEPROM
 */

#include <EEPROM.h> // for writing to storage space on chip

const byte bPressesInCode = 3; //How many knocks constitute a knock code
const byte PinButton = 4;
const byte PoutBlueLED = 6;

byte flag; //value read from EEPROM address0. 0 value will imply new board
unsigned int pressCode[bPressesInCode][2];  //1st element Button Press
                                           //2nd elemt is duration until next press
                                           //this is a Global Variable to be manipulated by subsequent functions

void Recalibrate(byte codeExists);
void SetPressCode();
void StoreCode();

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////Setup()
void setup() {
  
  pinMode(PinButton, INPUT);
  pinMode(PoutBlueLED, OUTPUT);
  digitalWrite(PoutBlueLED, LOW);

  Serial.begin(9600);
 
  //assign flag the value from EEPROM address 0, 
  EEPROM.write(0,0); 
  flag = EEPROM.read(0);
  if (flag == 0) //Should always be 0
  {
    Serial.println("SETUP");
    Recalibrate(flag);
  }
}

void loop() {
}

//////////////////////////////////////////////////////////////////////////////////////////////////Recalibrate() - used to set a new code/////////////////////////////////////////
void Recalibrate(byte codeExists) //codeExists is our flag value
{
  if (codeExists != 1) 
  {
    Serial.println("RECALIBRATING");
    SetPressCode();
    StoreCode();
    Serial.println("CODE STORED!");
  }
  else
  Serial.println("ERROR: Flag is NOT 0!");
}

/////////////////////////////////////////////////////////////////////////////////////////////////////////////SetPressCode()///////////////////////////////////////////////////////////////////
void SetPressCode()
{
  //Serial.println("SETLOCK");
  byte pressCounter = 0;
  int buttonPress=0; 
  int pressTime=0;
  unsigned int nSequenceBegin; //This will mark the time point where the button is first pressed
  digitalWrite(PoutBlueLED, HIGH);

  while (pressCounter < bPressesInCode) //make sure the maximum presses are not exceeded
  {
    buttonPress = digitalRead(PinButton);
    Serial.print("       Button: "); Serial.println(buttonPress);
    Serial.print("Press Counter: "); Serial.println(pressCounter); Serial.println();
    //XXXXXXX-- THIS IS WHERE THINGS BECOME IRRADIC --XXXXXXXXXXXX
    //XXXXXX-- If buton is held down, the entire sketch writes gibberish to the serial monitor and Serial.print() restarts from setup() --XXXXXXXXX
    //XXXXXX-- If I add a delay and release the button during the delay it works as expected --XXXXXXXXX
        
    while(buttonPress > 0) //XXXXXXXXXX-- This while loop never executes while button is held. --XXXXXXXXXXXXXXXX
    {
      if (pressCounter == 0)
      {
        pressCode[pressCounter][0] = pressCounter;
        pressCode[pressCounter][1] = 0; //may need to calibrate this for the time it takes to get to this line from analogRead(PinPiezo) i.e. minus 1, or 2, or 3...
        nSequenceBegin = millis();
      }
      else
      {
        pressCode[pressCounter][0] = pressCounter;
        pressCode[pressCounter][1] = millis() - nSequenceBegin;
      }
      pressCounter++;
      Serial.println(pressCounter);
      buttonPress = digitalRead(PinButton); 
    }
  }
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////StoreCode()////////////////////////////////////////////////////////////////////////////////
void StoreCode()
{
  //Serial.println("STORECODE");
  //DO NOT START AT 0 OTHERWISE OVERWRITE FLAG!!!
  int jjj = 0;
  for(int iii = 1; iii < (bPressesInCode*4) + 1; iii = iii+4) //require 4 bytes/knock (2 for knock intensity, 2 for time)
  {
    //knock into bytes:
    EEPROM.write(iii, (pressCode[jjj][0] >> 8)); 
    EEPROM.write(iii+1, pressCode[jjj][0]);
    //time into bytes:
    EEPROM.write(iii+2, (pressCode[jjj][1] >> 8));
    EEPROM.write(iii+3, pressCode[jjj][1]);
    jjj++;
    Serial.print("CODE STORED: "); Serial.println(iii);
   }

   digitalWrite(PoutBlueLED, LOW);
}
  
  //1. index &EEPROM at 1 (don't overwrite the flag)
  //2. Covert knockCode[][] from ints into logical bytes
  //3. Store each conversion in appropriate memory address then increment index

`

matthijskooijman, 2017-01-22:
Looking at your problem description, I strongly suspect this is a problem in your sketch code or logic, not in the Arduino IDE. Your summary talks about overwriting the stack, but your description does not indicate why you suspect this. Your example is way to complex to prove such behaviour, so you'll have to reduce it further to be useful.

I think your best approach here is to find help debugging your sketch. This is not the place for that, you should try the forums or IRC channel. I did look a bit into your sketch, one thing that springs to mind is that when the button is pressed, the `while(buttonPress > 0)` loop does not run so, neither does `pressCounter++;`, so `while (pressCounter < bPressesInCode)` also never becomes false. Not sure if this is intentional or not, though. I can't see anything in the sketch off-hand that would explain gibberish on the serial terminal.

In any case, I'm closing this report. If you do think there is an Arduino bug here, please reopen and provide further details.

closed 2017-01-22 by matthijskooijman

oqibidipo, 2017-01-24:
> `  while(buttonPress > 0) //XXXXXXXXXX-- This while loop never executes while button is held. --`

Not never - let's rephrase that loop:
```
while (button is pressed) 
{
    pressCode[pressCounter][*] = something;   // int [3][2] array
    pressCounter++; // 0..255 - quickly out of bounds unless you release the button *very* fast
    Serial.println(pressCounter);
}
```
That clobbers whatever is in the next 500 bytes after `pressCode` and that happens to include `Serial` with it's buffers and pointers to I/O registers.

matthijskooijman, 2017-01-24:
Ah, indeed. I was assuming a pullup and button to ground, but it's the reverse. As @oqibidipo points out, the inner loop writes outside of the `pressCode` array, which explains the problems.