arduino/Arduino #5902: Add new library for decoding JPEG images and rendering on a TFT

Bodmer, 2017-01-23:
https://github.com/Bodmer/JPEGDecoder

agdl, 2017-01-23:
I'm sorry, but your library can't be added since a tag is missing. Furthermore will be nice to move the source files in the src folder that is missing too. Please refer to libraries like this to see how libraries are made https://github.com/arduino-libraries/WiFi101

Bodmer, 2017-01-24:
OK, thanks I will study the WiFi101 example and update the repository.

Bodmer, 2017-01-24:
OK, I think it is now compliant with the required format.

shiftleftplusone, 2017-01-25:
hello,
CMIIW, but IMO some information or documentation is unfortunately missing in your lib:
For which displays will it work and for which not?
although you claim
" Adapting the example sketch for other TFT drivers and their graphics libraries should be quite easy if they support either setWindow() or SetAddrWindow() and pushColor() functions as found in the Adafruit_GFX library",
statements like "should be quite easy" is mostly misleading and  regularily fail if one tries to process this task (above all, if he/she is just a simple Arduino end user and even a starter, not a developer,or a  professional programmer).

So is the lib compatible to all existing alternative TFT Hardware SPI libs to display the jpgs, e.g. for ILI9340/1:
Adafruit_ILI9340 or Adafruit_ILI9341 lib plus Adafruit_GFX ,
and especially - above all - also to 
ILI9341_due:  https://github.com/marekburiak/ILI9341_due
( because of display processing speed reasons) ?

Bodmer, 2017-01-25:
Re: Info missing in the lib:
OK, so what you are suggesting is to add a tutorial inside the library (maybe as a PDF) or is the suggestion that there should be a documentation page somewhere on the web with an embedded link, for example in a Github Wiki?
There is an Adafruit_GFX + Adafruit_ILI9341 library example included with the library already so that point can be made clearer and essentially the library will work with any colour graphics display and any processor with sufficient memory resources, e.g. Mega, Due etc. that is compatible with the Arduino IDE.  Essentially the library is pretty hardware agnostic.

I will also add an example compatible with the stock IDE TFT library as that has already shown to be compatible.

I will cover those points and get back to you.

In the meantime, if you foresee any more blocking points then let me know.

Thanks.

shiftleftplusone, 2017-01-25:
hi,
my suggestion was both a compatibilty lsi for this lib and compatibility settings (#included? #defines?) to make the jpg libs work with already existing TFT libs (Adafruit, LIL9341_due, and perhaps also Henning Karlsen etc )

cmaglie, 2017-01-27:
I've added the library to the index.

Anyone interested may continue the discussion on the issue tracker of the library:
https://github.com/Bodmer/JPEGDecoder/issues

Thanks!

closed 2017-01-27 by cmaglie