arduino/Arduino #5904: Suggestion - IDE - Libraries

rogwabbit, 2017-01-23:
Could we add an external Library folder option, This allows upgrading of IDE to not delete existing separate user made or downloaded libraries.

per1234, 2017-01-23:
This feature already exists. It's called the sketchbook. The sketchbook location is set at **File > Preferences > Sketchbook location**. You should always install libraries to {sketchbook folder}/libraries. Don't ever install anything except the IDE to the Arduino IDE installation folder.

facchinm, 2017-01-24:
:+1:  for @per1234 explanation. The sketchbook does not get erased when you update the IDE and the libraries in there shadow the bundled ones. So if you need to modify a bundled library, simply copy its folder in the sketchbook, modify it and it will never be touched by the IDE

closed 2017-01-24 by facchinm