arduino/Arduino #5908: Missing Ethernet V2 (W5500) library based on current Arduino library 1.8.1 available

DieterBurandt, 2017-01-24:
Hi,
You don't offer a drive for the Arduino Ethernet Shield V2 that you offer on your web page.

I converted the library from https://github.com/adafruit/Ethernet2 to the Arduino Ethernet library 1.8.1 format including all updates and patches and added three enhancements: Options to set the SPI-speed and the CS pin and calls to read the Mac- and dhcpServerIP-address from the W5500. 

I did it using a file compare program. It does work perfect on my Arduino Due and is migrated to support (nearly) all devices that are supported by the Arduino team.

There is still a little work required. Things where I am not sure if it is worth to include and cannot test but most of the adaptions are regarding the correct copyright header, author list and to reach an agreement with the Arduino.org team to publish it as a product of Arduino.cc.

Except of the last points, it is worth to be a beta product and be published already from your side.

Due to the already implemented copyright headers (simple copied from your W5110 library) I don’t like to send it with this mail and make it public.

If you are interested send me a mail.

q2dg, 2017-01-27:
I think all these efforts (see for instance https://github.com/arduino-libraries/Ethernet/issues/37) should be centralized in some manner to be managed coherently  by Arduino Team developers to get finally a official good implementation, not many ones.

PaulStoffregen, 2018-07-27:
Fixed by Ethernet 2.0.0

closed 2018-07-27 by PaulStoffregen