arduino/Arduino #5910: add new library

msh2050, 2017-01-25:
wish to save time for others
https://github.com/msh2050/TimeCounter

septillion-git, 2017-01-25:
TimeCounter doesn't really cover what it does, does it? There is nothing "counter" about it...

msh2050, 2017-01-25:
thanks for review @septillion-git 
I made the counter outside the library "as in the example so the user will be free how to handle it".
For me I use tasks/timers to make the increment every second.
may by I should change the library name to secondsToHHMMSS.

septillion-git, 2017-01-25:
Yeah, I would indeed call it something like TimeFormater or something. That also leaves a lot of room for other nice formats.

I can also see a lot of improvements for the library ;) Issued you already one (which is a real bug) but I can go on 0:-)

msh2050, 2017-01-25:
thanks a lot .
I will change the name and try to commit the improvements.

cmaglie, 2017-01-27:
@msh2050 I'm closing the issue for now, when you're ready, please reopen the issue, or create another one with the new repository.

closed 2017-01-27 by cmaglie