arduino/Arduino #5914: POSIX Threads for Arduino

mikaelpatel, 2017-01-26:
There has been some questions about AVR implementation of the Scheduler. I have made available a portable Scheduler implementation which work on AVR, SAM, SAMD, Teensy, etc. https://github.com/mikaelpatel/Arduino-Scheduler.

There has also been a discussion about extending the Scheduler functionality. This has brought me to the conclusion that a possible way forward is to implement a sub-set of the POSIX Thread API for Arduino (pthreads). See https://github.com/mikaelpatel/Arduino-Scheduler/issues/13.

If there is an interest I could make time to do some work on this.

shiftleftplusone, 2017-01-26:
IMO that will be really incredibly awesome! 
(y)

closed 2017-02-15 by mikaelpatel

PaulStoffregen, 2017-02-15:
> I am really curious why not everybody round here is cheering for joy in anticipation of getting a reliable, worldwide tested and approved pre-emptive multithreading lib...

Just because you may have a "worldwide tested and approved pre-emptive multithreading lib" doesn't mean the huge ecosystem of other libraries will work in a multithreaded environment.  In fact, the vast majority use static data or other techniques that will randomly & intermittently fail if accessed from multiple preemptively scheduled threads.

Even if all those problems are solved, multi-threaded programming is difficult work.  Race conditions, deadlocks and other pitfalls await anyone trying to build their own inter-thread communication or synchronization code, even if using a well known API like pthreads.

mikaelpatel, 2017-02-15:
@PaulStoffregen Thanks for your comment. I forgot to motivate why I closed this issue/proposal. Your comment is very much to the point. I can only agree. 

The Scheduler (pthreads) risks opening a _can-of-worms_ too large to handle. The current Arduino core(s) needs refactoring before this becomes feasible.

shiftleftplusone, 2017-02-15:
I agree.
Unfortunately for Arduino there is missing any  reliably working multithreading lib ftm, some work not at all for AVRs, some have severe issues for the Due.
Perhaps this is really due to the fact that " the current Arduino core(s) needs refactoring".
Wth Posix pthread anyway I am already doing "amazing things", probably  because to me it's " in  the hands of the right people".