arduino/Arduino #5916: mrk zero freezes when using Scheduler

simplecast, 2017-01-27:
I have 2 new Arduino mkr 2 i was  looking for a way to do multi-tasking  i used this in my my code and tried it twice to confirm now both mkrzeros wont upload, won,t been seen by Arduino IDE but the computer sees it, i can't upload a empty sketch nothing everything i tried has failed

shiftleftplusone, 2017-01-27:
which Scheduler lib? there are at least 3 different ones, all named "Scheduler"

cmaglie, 2017-01-27:
> won't been seen by Arduino IDE but the computer sees it

Try this: push **quickly** the reset button two times, this should put the board into "bootloader" mode and allow you to upload again.

per1234, 2017-08-22:
@simplecast did you try the instructions that cmaglie provide? If so, did they solve the issue for you?

per1234, 2018-01-16:
Closing due to lack of response from @simplecast.

closed 2018-01-16 by per1234