arduino/Arduino #5918: arduino doesn't start after updating nvidia geforce driver to version 378.49

MaxBoyman, 2017-01-27:
Hi,
as a matter of fact this problem takes my two hole days from working.
the problem starts two days ago when i installed new nvidia geforce driver version 378.49.
my system config is:
Samsung NP300V5A-S03AE
and there is two graphic unit on my laptop:
1- intel HD 3000 onboard graphic 
2- nvidia gt 520 MX graphic
my operating system is windows 10 x64 home edition 14393.693 thanks to microsoft :)
everything were fine and i had the latest nvidia geforce driver before 378.49 and arduino were O.K.
but after updating every program were good but arduino.
arduino didn't start and after trying to run nothing were happend. 
but i am sure that arduino is running with my intel hd 3000 graphic processor.
for being sure on this problem. i soft reset. then hard reset, then re install from a disc 5 times with different types of program installation orders and i am now sure that there is some problem with this driver version and arduino ide.
i also tried different ide s 1.x.x , 1.6.x , 1.8.x, nightly hour builds, and then problem exists after installing this driver version. 
i am reporting to you to fix the bug.
ops i forgot that i also tried every FAQs you told, but nothing happened at all...

dlabun, 2017-01-28:
There are multiple reports on NVIDIA related forums with this driver breaking many other programs. I would recommend you downgrade to a different GeForce driver until NVIDIA straightens out this problem.

facchinm, 2017-01-30:
Hi @MaxBoyman ,
can you install 1.8.1 and execute `arduino_debug.exe` from a `cmd` shell (eventually reporting here the stacktrace)? It could be a problem with the bundled Java and it would help a lot if we could solve it before next major release.

MaxBoyman, 2017-01-30:
Hi @facchinm,
i installed the 1.8.1 version and executed arduino_debug.exe from cmd. but "windows problem reporting" told that there is a problem with java runtime. 
then it closes very fast with no further information. 
i should explains that i test this method in different situations:
1- i was installed all jre versions java 7 update 80 x86 and x64 and also java 8 update 121 x86 and x64 versions and problems were exists.
2- no jre installed at all and the problem also exists.
3- different orders of installations and installing in different selections of jre versions.
there were problems exists with no changes.
and believe me i tried these different situations with every time reinstalling windows from first step.

facchinm, 2017-01-30:
I believe you :slightly_smiling_face: The jvm we use is bundled with the IDE, so any external Java installation should make no difference. We are currently shipping jvm8u111, I'll search if there are bug reports about incompatibilities between newer Nvidia drivers and this Java version and report here.

facchinm, 2017-01-30:
Ok, it looks quite widespread https://www.reddit.com/r/feedthebeast/comments/5pydv9/dont_install_nvidia_gpu_driver_version_37849_it/

I hope Nvida will fix their own drivers soon to avoid breakage with much more Java based apps

facchinm, 2017-01-30:
Temporary solution [here](https://www.reddit.com/r/feedthebeast/comments/5pydv9/dont_install_nvidia_gpu_driver_version_37849_it/dcxav2w/)

MaxBoyman, 2017-01-30:
Thanks a lot. 
and about temporary solution, it seems to be worked with devices which supports Nvidia 3D Vision, but my device's Graphic Card GT520 MX was not included with 3d vision support in chip. so this solution not working for me. i was reinstalled windows and install WHQL microsoft Certified driver for my graphic card with windows update. thank to microsoft for forcing people to download and install their drivers via windos update. :)
so as i see, the problem is not with java bundle which you included in arduino 1.8.1.

MaxBoyman, 2017-01-30:
i am still waiting for problem solution and will comment here if any.

neurooctober, 2017-02-03:
Interesting, I've tried unpacking _arduino-1.0.6-windows.zip_ and it works flawlessly with 378.49 driver.

neurooctober, 2017-02-03:
GeForce Hot Fix driver version 378.57 worked for me: http://nvidia.custhelp.com/app/answers/detail/a_id/4378/

> -        Fixed crash in Minecraft and some other Java-based titles

MaxBoyman, 2017-02-03:
problem solved with the installation of latest version of Nvidia driver 378.57. thanks to @neurooctober!

facchinm, 2017-02-07:
Closing as solved upstream

closed 2017-02-07 by facchinm