arduino/Arduino #5919: cannot use sstream library

sanfx, 2017-01-27:
I am trying to use sstream library 

`#include <sstream>`

I want to achieve 

```
sqlQuery  << "INSERT INTO arduinoSensorData.sensorLog (out_temperature,  " 
        << "out_humidity,  drwngRoom_temperature, drwngRoom_humidity, "
        << "pot1_soilMoisture, pot1_avg_SoilMoisture, wateringPot1) VALUES ('" 
        << outdoorTempInC << "', '" << outoorHumidity << "', '" << indorTempinC
        << "', '" << humidity << "', '" << val << "', '" << avgVal << "', '"
        << statusMsg << "');";
```
but i do not want to use string.concat, or sprintf

facchinm, 2017-01-30:
Hi @sanfx ,
you should contact the library developer or write on the forum for these kind of questions, this repo is intended to collect issues for the IDE itself.
Thanks

closed 2017-01-30 by facchinm

matthijskooijman, 2017-01-31:
Actually, `<sstream>` is from the standard C++ library, but that library is not available on AVR-based Arduinos. There is a port of uclibc++ available that contains `sstream`, but I'm not sure if that part actually works on Arduino too. It's here if you want to try it: https://github.com/maniacbug/StandardCplusplus